electrified/asus-wmi-sensors #1: Strix B450-F might not be supported

jasondaigo, 2018-12-19:
I cannot get a output like u on my Strix motherboard. Might be not suitable for that model.
this is before the install and after removal of it87
```[jason@behemoth ~]$ sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

amdgpu-pci-0b00
Adapter: PCI adapter
vddgfx:       +0.75 V  
fan1:        1027 RPM
temp1:        +33.0°C  (crit = +91.0°C, hyst = -273.1°C)
power1:        3.00 W  (cap = 180.00 W)

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +41.0°C  (high = +70.0°C)
Tctl:         +41.0°C  
```
then i installed the aur package 

```[jason@behemoth ~]$ pikaur -S asus-wmi-sensors-dkms-git
Reading repository package databases...
Reading local package database...
Resolving AUR dependencies...

:: AUR package will be installed:
 asus-wmi-sensors-dkms-git                                  -> 6.04e5cc0-1

:: Proceed with installation? [Y/n] 
:: [v]iew package details   [m]anually select packages
>> 
looking for conflicting AUR packages...
:: warning: Not showing diff for asus-wmi-sensors-dkms-git package (already reviewed)
Do you want to edit PKGBUILD for asus-wmi-sensors-dkms-git package? [y/N] N

:: Downloading the latest sources for a devel package asus-wmi-sensors-dkms-git...

:: Starting the build:
==> Making package: asus-wmi-sensors-dkms-git 13.0b9285a-1 (Wed Dec 19 17:52:57 2018)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating asus-wmi-sensors git repo...
Fetching origin
  -> Found asus-wmi-sensors.conf
  -> Found dkms.conf
  -> Found module-load.conf
==> Validating source files with sha256sums...
    asus-wmi-sensors ... Skipped
    asus-wmi-sensors.conf ... Passed
    dkms.conf ... Passed
    module-load.conf ... Passed
==> Extracting sources...
  -> Creating working copy of asus-wmi-sensors git repo...
Switched to a new branch 'makepkg'
==> Starting prepare()...
==> Starting pkgver()...
==> Entering fakeroot environment...
==> Starting package()...
==> Tidying install...
  -> Removing libtool files...
  -> Purging unwanted files...
  -> Removing static library files...
  -> Stripping unneeded symbols from binaries and libraries...
  -> Compressing man and info pages...
==> Checking for packaging issues...
==> Creating package "asus-wmi-sensors-dkms-git"...
  -> Generating .PKGINFO file...
  -> Generating .BUILDINFO file...
  -> Generating .MTREE file...
  -> Compressing package...
==> Leaving fakeroot environment.
==> Finished making: asus-wmi-sensors-dkms-git 13.0b9285a-1 (Wed Dec 19 17:52:58 2018)

loading packages...
resolving dependencies...
looking for conflicting packages...

Packages (1) asus-wmi-sensors-dkms-git-13.0b9285a-1

Total Installed Size:  0.07 MiB

:: Proceed with installation? [Y/n] Y
(1/1) checking keys in keyring                                               [############################################] 100%
(1/1) checking package integrity                                             [############################################] 100%
(1/1) loading package files                                                  [############################################] 100%
(1/1) checking for file conflicts                                            [############################################] 100%
(1/1) checking available disk space                                          [############################################] 100%
:: Processing package changes...
(1/1) installing asus-wmi-sensors-dkms-git                                   [############################################] 100%
:: Running post-transaction hooks...
(1/2) Install DKMS modules
==> dkms install asus-wmi-sensors/13.0b9285a -k 4.19.9-arch1-1-ARCH
(2/2) Arming ConditionNeedsUpdate...
[jason@behemoth ~]$ 
```
and after a reboot i get this output (the asus one now at 3rd position.
```[jason@behemoth ~]$ sensors
k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +31.8°C  (high = +70.0°C)
Tctl:         +31.8°C  

amdgpu-pci-0b00
Adapter: PCI adapter
vddgfx:       +0.75 V  
fan1:        1032 RPM
temp1:        +36.0°C  (crit = +91.0°C, hyst = -273.1°C)
power1:        3.00 W  (cap = 180.00 W)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM
```
modinfo
```[jason@behemoth ~]$ sudo modinfo asus_wmi_sensors
filename:       /lib/modules/4.19.9-arch1-1-ARCH/extramodules/kernel/drivers/hwmon/asus-wmi-sensors.ko
license:        GPL
description:    Asus WMI Sensors Driver
author:         Ed Brindley <kernel@maidavale.org>
srcversion:     99760DC6A51239BE5C30419
depends:        wmi
name:           asus_wmi_sensors
vermagic:       4.19.9-arch1-1-ARCH SMP preempt mod_unload modversions 
```
i have no access to a windows os. not even pirated one :-) 
cant confirm whether its WMI or not

jasondaigo, 2018-12-20:
> If you could send me that output please.
 i will.ty

jasondaigo, 2018-12-20:
this is the output
```[jason@behemoth ~]$ dmesg | grep asus
[    3.878086] asus_wmi_sensors: loading out-of-tree module taints kernel.
[    3.878099] asus_wmi_sensors: module verification failed: signature and/or required key missing - tainting kernel
[    3.878222] asuswmisensors: WMI GUID matched - probing
[    3.878369] asuswmisensors: Vendor: ASUSTeK COMPUTER INC. Board: ROG STRIX B450-F GAMING BIOS version: 1103 WMI version: 0
[    3.878369] asuswmisensors: Unsupported board
[    3.977261] asus_wmi: ASUS WMI generic driver loaded
[    3.978425] asus_wmi: Initialization: 0x0
[    3.978453] asus_wmi: BIOS WMI version: 0.9
[    3.978498] asus_wmi: SFUN value: 0x0
[    3.979033] asus_wmi: Number of fans: 1
```

electrified, 2018-12-26:
Many thanks for the output. My driver only supports  ASUSHW WMI versions 2 and 3 due to locking issues with earlier versions, and your board is reporting version 0.

For the time being I will update the README to indicate that your board isn't supported, and I will try and find out what workarounds HWiNFO is using to get round the locking issues.

jasondaigo, 2018-12-26:
since then i enjoy the old stuff. :-) happy holidays

closed 2018-12-26 by jasondaigo