electrified/asus-wmi-sensors #2: Does not seem to work on Crosshair VII Hero WiFi

evilphish, 2019-01-02:
Hey there. Thank you so much for attempting to support our boards. However, I can't seem to get your module working. It compiles and installs just fine. It also loads fine:

```
[19739.536807] asuswmisensors: WMI GUID matched - probing
[19739.537227] asuswmisensors: Vendor: ASUSTeK COMPUTER INC. Board: ROG CROSSHAIR VII HERO (WI-FI) BIOS version: 0804 WMI version: 1
```
however:
```
$> sensors
k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +40.5°C  (high = +70.0°C)
Tctl:         +50.5°C  

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM
```

no sign of the virtual asus device. The module seems to be loaded fine:
```
$> lsmod | grep asus
asus_wmi_sensors       20480  0
asus_wmi               28672  1 eeepc_wmi
sparse_keymap          16384  1 asus_wmi
video                  49152  1 asus_wmi
wmi                    28672  4 asus_wmi_sensors,asus_wmi,wmi_bmof,mxm_wmi
```

System is Linux Mint 19.1 running the 4.20 kernel. If you need more info, please don't hesitate to ask :)

electrified, 2019-01-03:
Hi,

If you upgrade your BIOS to 1002 or 1103 it should work.

The older WMI version 1 in 0804 is not supported due to locking issues.

I will update the readme to explain this. I'll have a look at why the
module loaded too, as it shouldn't have done with the WMI version mismatch.

Thanks,
Ed

<https://github.com/electrified/asus-wmi-sensors/issues/2>, or mute the
thread

evilphish, 2019-01-03:
Thanks for your quick reply. I did the BIOS update to 1103 but I can still not see any sensor output of your virtual device. The logs now read

```
[   30.227936] asuswmisensors: WMI GUID matched - probing
[   30.228345] asuswmisensors: Vendor: ASUSTeK COMPUTER INC. Board: ROG CROSSHAIR VII HERO (WI-FI) BIOS version: 1103 WMI version: 2
```

```
$> sensors
k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +42.0°C  (high = +70.0°C)
Tctl:         +52.0°C  

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM
```

electrified, 2019-01-03:
Hi there,

I think the version of lm_sensors in Mint/Ubuntu may be the problem now, as I rely on a fix that has only gone into the latest 3.5.0 version.

What does `ls -l /sys/bus/wmi/devices/466747A0-70EC-11DE-8A39-0800200C9A66/hwmon/hwmon*/` return with the driver loaded?

There should be a load of _input and _label files. You can get the raw sensor values and labels by catting these.

If that's the case then the driver is working, but lm_sensors 3.4.0 sensors is discarding the device and upgrading to lm_sensors 3.5.0 should get it working.

electrified, 2019-01-03:
The change in lm_sensors 3.5.0 which allows my WMI bus device to show up as on the virtual but is this one: https://github.com/lm-sensors/lm-sensors/commit/7e976f1325e55b5e26a63c8f9c7af0cb9aac3aa7

evilphish, 2019-01-03:
```
$> ls -l /sys/bus/wmi/devices/466747A0-70EC-11DE-8A39-0800200C9A66/hwmon/hwmon*/
total 0
-r--r--r-- 1 root root 4096 Jan  3 21:39 curr1_input
-r--r--r-- 1 root root 4096 Jan  3 21:39 curr1_label
lrwxrwxrwx 1 root root    0 Jan  3 21:38 device -> ../../../466747A0-70EC-11DE-8A39-0800200C9A66
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan1_input
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan1_label
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan2_input
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan2_label
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan3_input
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan3_label
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan4_input
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan4_label
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan5_input
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan5_label
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan6_input
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan6_label
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan7_input
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan7_label
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan8_input
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan8_label
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan9_input
-r--r--r-- 1 root root 4096 Jan  3 21:38 fan9_label
-r--r--r-- 1 root root 4096 Jan  3 21:39 in0_input
-r--r--r-- 1 root root 4096 Jan  3 21:39 in0_label
-r--r--r-- 1 root root 4096 Jan  3 21:39 in10_input
-r--r--r-- 1 root root 4096 Jan  3 21:39 in10_label
-r--r--r-- 1 root root 4096 Jan  3 21:39 in11_input
-r--r--r-- 1 root root 4096 Jan  3 21:39 in11_label
-r--r--r-- 1 root root 4096 Jan  3 21:39 in12_input
-r--r--r-- 1 root root 4096 Jan  3 21:39 in12_label
-r--r--r-- 1 root root 4096 Jan  3 21:39 in13_input
-r--r--r-- 1 root root 4096 Jan  3 21:39 in13_label
-r--r--r-- 1 root root 4096 Jan  3 21:39 in1_input
-r--r--r-- 1 root root 4096 Jan  3 21:39 in1_label
-r--r--r-- 1 root root 4096 Jan  3 21:39 in2_input
-r--r--r-- 1 root root 4096 Jan  3 21:39 in2_label
-r--r--r-- 1 root root 4096 Jan  3 21:39 in3_input
-r--r--r-- 1 root root 4096 Jan  3 21:39 in3_label
-r--r--r-- 1 root root 4096 Jan  3 21:39 in4_input
-r--r--r-- 1 root root 4096 Jan  3 21:39 in4_label
-r--r--r-- 1 root root 4096 Jan  3 21:39 in5_input
-r--r--r-- 1 root root 4096 Jan  3 21:39 in5_label
-r--r--r-- 1 root root 4096 Jan  3 21:39 in6_input
-r--r--r-- 1 root root 4096 Jan  3 21:39 in6_label
-r--r--r-- 1 root root 4096 Jan  3 21:39 in7_input
-r--r--r-- 1 root root 4096 Jan  3 21:39 in7_label
-r--r--r-- 1 root root 4096 Jan  3 21:39 in8_input
-r--r--r-- 1 root root 4096 Jan  3 21:39 in8_label
-r--r--r-- 1 root root 4096 Jan  3 21:39 in9_input
-r--r--r-- 1 root root 4096 Jan  3 21:39 in9_label
-r--r--r-- 1 root root 4096 Jan  3 21:38 name
drwxr-xr-x 2 root root    0 Jan  3 21:39 power
lrwxrwxrwx 1 root root    0 Jan  3 21:39 subsystem -> ../../../../../../../../class/hwmon
-r--r--r-- 1 root root 4096 Jan  3 21:38 temp1_input
-r--r--r-- 1 root root 4096 Jan  3 21:38 temp1_label
-r--r--r-- 1 root root 4096 Jan  3 21:38 temp2_input
-r--r--r-- 1 root root 4096 Jan  3 21:38 temp2_label
-r--r--r-- 1 root root 4096 Jan  3 21:38 temp3_input
-r--r--r-- 1 root root 4096 Jan  3 21:38 temp3_label
-r--r--r-- 1 root root 4096 Jan  3 21:38 temp4_input
-r--r--r-- 1 root root 4096 Jan  3 21:38 temp4_label
-r--r--r-- 1 root root 4096 Jan  3 21:38 temp5_input
-r--r--r-- 1 root root 4096 Jan  3 21:38 temp5_label
-r--r--r-- 1 root root 4096 Jan  3 21:38 temp6_input
-r--r--r-- 1 root root 4096 Jan  3 21:38 temp6_label
-r--r--r-- 1 root root 4096 Jan  3 21:38 temp7_input
-r--r--r-- 1 root root 4096 Jan  3 21:38 temp7_label
-r--r--r-- 1 root root 4096 Jan  3 21:38 temp8_input
-r--r--r-- 1 root root 4096 Jan  3 21:38 temp8_label
-rw-r--r-- 1 root root 4096 Jan  3 21:38 uevent
```

cating them results in:
```
 cat *
4000
CPU VRM Output Current
cat: device: Is a directory
2057
CPU Fan
670
Chassis Fan 1
661
Chassis Fan 2
663
Chassis Fan 3
0
HAMP Fan
0
Water Pump
0
CPU OPT
0
Water Flow
0
AIO Pump
1341
CPU Core Voltage
1094
SB 1.05V Voltage
1351
CPU Core Voltage
1136
CPU SOC Voltage
1400
DRAM Voltage
1123
CPU SOC Voltage
1384
DRAM Voltage
556
VDDP Voltage
1984
1.8V PLL Voltage
12154
+12V Voltage
4960
+5V Voltage
3314
3VSB Voltage
3205
VBAT Voltage
3314
AVCC3 Voltage
asuswmisensors
cat: power: Is a directory
cat: subsystem: Is a directory
31000
CPU Temperature
33000
CPU Socket Temperature
28000
Motherboard Temperature
44000
Chipset Temperature
216000
Tsensor 1 Temperature
38000
CPU VRM Temperature
216000
Water In
216000
Water Out
```

So it seems to work, even though I find the super large values at the very bottom (i.e. VRM Temperature) confusing ^^

I have to check and see where I can get a newer lm_sensors then.

electrified, 2019-01-03:
Great stuff!

Yes, sysfs returns large integers and these are scaled within libsensors to the actual decimal value.

Voltages and temps need to be divided by 1000 to get real value.

The super high values of 216degC indicates no sensor is connected. I am thinking of filtering these out and just returning 0 deg. You can see the same behaviour in some Windows apps such as HWiNFO so I haven't bothered so far.

evilphish, 2019-01-03:
Awesome, that explains a lot. I have also found out that there is no easy way to get lm-sensors 5 onto my system as it conflicts with libsensors4 which is in turn required by a multitude of packages, culminating in what is about half of my desktop environment :)

Too bad really, most of the data that can be read via your module is super informative :/ Guess I will have to wait for my system to get up to libsensors5 on its own. Thanks for your help in getting to the bottom of this though...

electrified, 2019-01-03:
Hi,

I've made some changes to the driver in this commit https://github.com/electrified/asus-wmi-sensors/commit/5a813481e008fc1afaac3eb50aa78f400d4792e3, so it's now registered on the platform bus rather than the WMI bus and the hwmon entries show up here: `/sys/bus/platform/devices/asus-wmi-sensors/hwmon/hwmon*/`

I think this should enable it to work with lm_sensors 3.4.0.

Can you give it a go and see if it works?

evilphish, 2019-01-04:
You sir, are awesome :)

```
$> sensors
asuswmisensors-isa-0000
Adapter: ISA adapter
CPU Core Voltage:         +1.34 V  
CPU SOC Voltage:          +1.12 V  
DRAM Voltage:             +1.38 V  
VDDP Voltage:             +0.57 V  
1.8V PLL Voltage:         +1.98 V  
+12V Voltage:            +12.15 V  
+5V Voltage:              +4.96 V  
3VSB Voltage:             +3.29 V  
VBAT Voltage:             +3.16 V  
AVCC3 Voltage:            +3.31 V  
SB 1.05V Voltage:         +1.10 V  
CPU Core Voltage:         +1.35 V  
CPU SOC Voltage:          +1.14 V  
DRAM Voltage:             +1.40 V  
CPU Fan:                 2064 RPM
Chassis Fan 1:            695 RPM
Chassis Fan 2:            675 RPM
Chassis Fan 3:            674 RPM
HAMP Fan:                   0 RPM
Water Pump:                 0 RPM
CPU OPT:                    0 RPM
Water Flow:                 0 RPM
AIO Pump:                   0 RPM
CPU Temperature:          +43.0°C  
CPU Socket Temperature:   +34.0°C  
Motherboard Temperature:  +29.0°C  
Chipset Temperature:      +45.0°C  
Tsensor 1 Temperature:   +216.0°C  
CPU VRM Temperature:      +40.0°C  
Water In:                +216.0°C  
Water Out:               +216.0°C  
CPU VRM Output Current:   +9.00 A  
```

electrified, 2019-01-04:
Great, glad it's working for you.

I'll close this issue now, let me know if you have any further problems.

closed 2019-01-04 by electrified