electrified/asus-wmi-sensors #3: Error on make dkms

KeithMyers, 2019-01-18:
Just tried this new driver but hit a make snag.  The terminal output.
```
keith@Serenity:~/asus-wmi-sensors$ sudo make dkms
[sudo] password for keith: 

Creating symlink /var/lib/dkms/asus-wmi-sensors/78911e0/source ->
                 /usr/src/asus-wmi-sensors-78911e0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j16 KERNELRELEASE=4.15.0-43-generic TARGET=4.15.0-43-generic...(bad exit status: 2)
ERROR (dkms apport): binary package for asus-wmi-sensors: 78911e0 not found
Error! Bad return status for module build on kernel: 4.15.0-43-generic (x86_64)
Consult /var/lib/dkms/asus-wmi-sensors/78911e0/build/make.log for more information.
Makefile:91: recipe for target 'dkms' failed
make: *** [dkms] Error 10
```
And the make.log output

`DKMS make.log for asus-wmi-sensors-78911e0 for kernel 4.15.0-43-generic (x86_64)
Thu Jan 17 23:57:09 PST 2019
  CC [M]  /var/lib/dkms/asus-wmi-sensors/78911e0/build/asus-wmi-sensors.o
/var/lib/dkms/asus-wmi-sensors/78911e0/build/asus-wmi-sensors.c:113:43: error: ‘hwmon_max’ undeclared here (not in a function); did you mean ‘hwmon_fan’?
  const struct asus_wmi_sensor_info **info[hwmon_max];
                                           ^~~~~~~~~
                                           hwmon_fan
/var/lib/dkms/asus-wmi-sensors/78911e0/build/asus-wmi-sensors.c: In function ‘configure_sensor_setup’:
/var/lib/dkms/asus-wmi-sensors/78911e0/build/asus-wmi-sensors.c:447:25: warning: left-hand operand of comma expression has no effect [-Wunused-value]
   nr_count[hwmon_chip]++, nr_types++;
                         ^
/var/lib/dkms/asus-wmi-sensors/78911e0/build/asus-wmi-sensors.c:399:6: warning: unused variable ‘nr_count’ [-Wunused-variable]
  int nr_count[hwmon_max] = {0}, nr_types = 0;
      ^~~~~~~~
scripts/Makefile.build:339: recipe for target '/var/lib/dkms/asus-wmi-sensors/78911e0/build/asus-wmi-sensors.o' failed
make[3]: *** [/var/lib/dkms/asus-wmi-sensors/78911e0/build/asus-wmi-sensors.o] Error 1
Makefile:1551: recipe for target '_module_/var/lib/dkms/asus-wmi-sensors/78911e0/build' failed
make[2]: *** [_module_/var/lib/dkms/asus-wmi-sensors/78911e0/build] Error 2
Makefile:72: recipe for target 'modules' failed
make[1]: *** [modules] Error 2`

electrified, 2019-01-18:
Hi, this is due to kernel 4.15 not having `hwmon_max` defined.

It looks like it was first added in 4.17 https://github.com/torvalds/linux/commit/d57538004b2e57be6a5d8583b65d1b049245abf7

Should be easy to work around, I'll try and fix it in the next few days.

KeithMyers, 2019-01-18:
Just a FYI your new driver installed perfectly on my test Ubuntu 18.10 partition with kernel 4.18.13. 

All of the WMI sensors in BIOS 1002 on my ASUS Crosshair VII Hero board exported to sensors.  Now have lovely sensor monitors added to GKrellm Monitor on the Desktop.

Thanks a bunch.  Eagerly waiting the new version for the earlier kernels in Ubuntu 18.04

KeithMyers, 2019-01-19:
Yes, it builds and installs on Ubuntu 18.04 with kernel 4.15.0-43 now.  Works fine and shows everything that it showed on Ubuntu 18.10 with kernel 4.18.0-13 that I first tried it on.

Thanks for such a fast solution.  Greatly appreciated by me and the whole Linux community.  I have posted of the new version in the BOINC Seti forums for the crunchers and also in the Overclock.net Crosshair VII Hero forum.

electrified, 2019-01-19:
Great, glad it is now working for you and thanks for spreading the word!

closed 2019-01-19 by electrified