electrified/asus-wmi-sensors #6: Zenith Extreme

Ljugis, 2019-01-28:
Hi
The asus-wmi-sensors module don't seem to work on Zenith Extreme, UEFI 1601 modified by Elmor. 

after modprobe asus-wmi-sensors:
~$ sensors
asuswmisensors-isa-0000
Adapter: ISA adapter

And it gets stuck there.
If I try "modprobe -r asus-wmi-sensors" it also get stuck.
When I tried to run "sensors" before loading this module, I got some information.
HWinfo in Windows reports information using ASUSWMI as it should.

$ uname -a
Linux XXXXX 4.18.0-13-generic #14-Ubuntu SMP Wed Dec 5 09:04:24 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

electrified, 2019-01-28:
Hi, thanks for trying it.

Can you post any output you get in `dmesg` after running `sensors`?

As you can't remove the module I'm guessing there has been a kernel oops, so there should be some stack trace etc.

Thanks.

thagabe, 2019-01-28:
Doesn't work on Arch with Linux Kernel 4.20.4 either. Same issue, will provide dmesg output as soon as I get home.

Ljugis, 2019-01-29:
output of dmesg:

~$ dmesg
[    0.000000] Linux version 4.18.0-13-generic (buildd@lgw01-amd64-048) (gcc version 8.2.0 (Ubuntu 8.2.0-7ubuntu1)) #14-Ubuntu SMP Wed Dec 5 09:04:24 UTC 2018 (Ubuntu 4.18.0-13.14-generic 4.18.17)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.18.0-13-generic root=UUID=8072703a-aabc-4fab-95e2-e2a6ae2b2dcc ro quiet splash vt.handoff=1
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009deffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009df0000-0x0000000009ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000a000000-0x000000000affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000b000000-0x000000000b01ffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000b020000-0x0000000076807fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000076808000-0x0000000077d4efff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000077d4f000-0x0000000077d7dfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000077d7e000-0x0000000078243fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000078244000-0x000000007952dfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007952e000-0x00000000795e3fff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000795e4000-0x000000007bffffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007c000000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b7a00000-0x00000000b7a00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b7b00000-0x00000000b7b7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ef800000-0x00000000ef8fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000efc00000-0x00000000efc00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000efd00000-0x00000000efd7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fea00000-0x00000000feafffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec30000-0x00000000fec30fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedc0000-0x00000000fedc0fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedc2000-0x00000000fedcffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedd4000-0x00000000fedd5fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000feefffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000107f2fffff] usable
[    0.000000] BIOS-e820: [mem 0x000000107f300000-0x000000107fffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.60 by American Megatrends
[    0.000000] efi:  ACPI 2.0=0x77d58000  ACPI=0x77d58000  SMBIOS=0x7949c000  SMBIOS 3.0=0x7949b000  MPS=0xfca80  ESRT=0x71cc2f18  MEMATTR=0x71a2d018 
[    0.000000] secureboot: Secure boot could not be determined (mode 0)
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] DMI: System manufacturer System Product Name/ROG ZENITH EXTREME, BIOS 1601 ASUSHW 10/26/2018
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] last_pfn = 0x107f300 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000000 mask FFFF80000000 write-back
[    0.000000]   1 base 00007C000000 mask FFFFFC000000 uncachable
[    0.000000]   2 disabled
[    0.000000]   3 disabled
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[... 1644 lines cut ...]
[ 1345.968251]  do_syscall_64+0x5a/0x110
[ 1345.968255]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[ 1345.968257] RIP: 0033:0x7f834611df31
[ 1345.968258] Code: fe ff ff 50 48 8d 3d 46 6e 0a 00 e8 59 35 02 00 66 0f 1f 84 00 00 00 00 00 48 8d 05 c9 f7 0d 00 8b 00 85 c0 75 13 31 c0 0f 05 <48> 3d 00 f0 ff ff 77 57 c3 66 0f 1f 44 00 00 41 54 49 89 d4 55 48 
[ 1345.968281] RSP: 002b:00007ffcf2bb85a8 EFLAGS: 00000246 ORIG_RAX: 0000000000000000
[ 1345.968283] RAX: ffffffffffffffda RBX: 000055f641834330 RCX: 00007f834611df31
[ 1345.968285] RDX: 0000000000001000 RSI: 000055f64182cf90 RDI: 0000000000000003
[ 1345.968286] RBP: 0000000000000d68 R08: 000055f64182cf80 R09: 0000000000000003
[... cut ...]
[ 1345.968380] RDX: 0000000000000000 RSI: ffffffffa7891e90 RDI: 0000000000000000
[ 1345.968381] RBP: ffffa5e51290fd60 R08: 0000000000000000 R09: ffffffffa6fad857
[ 1345.968382] R10: ffffe1a3c09a4740 R11: 0000000000000001 R12: ffff955df01b1c18
[ 1345.968384] R13: 0000000000000001 R14: ffffa5e51290fd70 R15: 0000000000000022
[ 1345.968386] FS:  00007f8345e84740(0000) GS:ffff955dfd000000(0000) knlGS:0000000000000000
[ 1345.968387] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1345.968389] CR2: 000000000000002c CR3: 0000001006ed6000 CR4: 00000000003406e0

electrified, 2019-01-29:
Thanks.

Could you get a screenshot of HWiNFO, and also the raw output from SIV (see last screenshot in this comment: https://github.com/electrified/asus-wmi-sensors/issues/4#issuecomment-456171554) as they may be helpful.

Ljugis, 2019-01-29:
[image]
[image]
Attached screenshots

electrified, 2019-02-03:
Hi @Ljugis I've made a change which I think may fix it - obviously it is difficult to test without the board myself :D.

If you could give it a go and let me know that would be great.

Ljugis, 2019-02-04:
It works now with the latest update :-)

[image]

closed 2019-02-04 by Ljugis