electrified/asus-wmi-sensors #8: AUR Update

thagabe, 2019-02-05:
Board: Zenith Extreme
Checked the code and it works correctly on Arch Linux. I was just checking to see if you could update the AUR package builder! 

Thanks for the great code!

electrified, 2019-02-05:
Hi there!

The AUR package pulls from git head, so should always get the latest code.

However I have seen issues around DKMS not replacing the module when upgrading the package (I thought this issue had been resolved giving the module an explicit version number rather than letting it default to the GIT hash).

Removing the package, removing the module via the DKMS command and then reinstalling has sorted it in the past.

My PKGBUILD was based off other DKMS kernel module PKGBUILDs, but definitely not an expert, so open to any suggestions on improvements... :)

thagabe, 2019-02-05:
I'm too new to the AUR to be effective on suggesting improvements. However, when updating using 'yay' the wmi did not show as having updates (not sure how they determine updates), looking into the PKGBUILD I noticed line 5 "pkgver=6.04e5cc0" the alphanumeric characters "04e5cc0" seemed to refer to a previous commit.

Now that i see the code again, it does look like it pulls from the latest source "git rev-list --count HEAD" which is good for a first time install but does seem to hinder automatic updates through an AUR wrapper as it doesn't know there has been commits. 

I opted to edit the PKGBUILD to reflect the pkgver of the latest commit and rebuilt. Got the latest code and after a reboot (dkms did not remove the previous module and inserting the new one led to terminal "freeze" [ctrl+c did not kill the signal]) it worked as expected!

electrified, 2019-02-05:
I'll keep this ticket open and try and improve the situation!

My PKGFILE was based on this one https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=it87-dkms-git

I'll give the exfat AUR package a go and see if that reliably can be updated, as the underlying code is fairly frequently updated, but the AUR package not https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=exfat-dkms-git)

My gut feel is that updating the AUR packaging every time I update the main git repo shouldn't be necessary with most AUR wrappers, but I may have made an error with my packaging :D. In particular `yay`s FAQ suggests git package updates are handled with a hash cache: https://github.com/Jguer/yay/#i-know-my--git-package-has-updates-but-yay-doesnt-offer-to-update-it

thagabe, 2019-04-20:
Just got the Zenith Extreme Alpha and not working on this versions of the zenith. Please let me know if I can help out in debug!

electrified, 2019-04-22:
Hi, as per my comment on #15, I've pushed a new version including the DMI board name for the Alpha. 

Regarding this ticket about the AUR packaging, I am going to close it as I believe it is working correctly:

After pushing updates to the repo, and running ```yay -Syu --devel```, the package is rebuilt

```
130 ed@zoomer ~/dev/github/asus-wmi-sensors (git)-[master] % yay -Syu --devel                                                                                      :(
:: Synchronising package databases...
 core is up to date
 extra is up to date
 community is up to date
 multilib is up to date
:: Starting full system upgrade...
 there is nothing to do
:: Searching databases for updates...
:: Searching AUR for updates...
:: Checking development packages...
 -> Out Of Date AUR Packages:  opencl-amd
:: 2 Packages to upgrade.
2  devel/asus-wmi-sensors-dkms-git  24.fe0ac60-1        -> latest-commit
1  devel/minipro-git                0.1.r159.g57b293d-1 -> latest-commit
.....output truncated
```

Running `yay -Syu --devel` a second time, the package is now not rebuilt as there are no changes to the git tree:

```
ed@zoomer ~/dev/github/asus-wmi-sensors (git)-[master] % yay -Syu --devel
:: Synchronising package databases...
 core is up to date
 extra is up to date
 community is up to date
 multilib is up to date
:: Starting full system upgrade...
 there is nothing to do
:: Searching databases for updates...
:: Searching AUR for updates...
:: Checking development packages...
 -> Out Of Date AUR Packages:  opencl-amd
 there is nothing to do
```

closed 2019-04-22 by electrified