electrified/asus-wmi-sensors #10: ASUS Zenith Extreme w/ BIOS 1607 working

KeithMyers, 2019-03-08:
I helped a fellow BOINC user get up and running the asus-wmi-sensors driver on his ASUS ROG Zenith Extreme motherboard.  I asked him to post his sensors output but he was unfamiliar with the workings of github, created his own repository to post the text file.  I will put his repository link here.

[https://github.com/captainiom/sensors-asus-extreme-zenith-rog/blob/master/sensor.txt](url)
He and I questioned the temp sensors readings with the 216° C. on all the outputs.  I explained the unterminated reading is normal.  But I wonder about the DIMM 2 readings of 0° C. being different than that.  I don't believe he has any temp sensors installed and he is unsure as he did not build the machine, a system builder did.

This was just a note that the ASUS ROG Zenith Extreme is a known working motherboard with the drivers as long as the BIOS is updated to 1607.

electrified, 2019-03-19:
Thanks for the heads up - I will update the README to indicate the Elmor hacked BIOS is no longer needed.

captainiom, 2019-03-20:
An analysis of what is right and what is wrong from the developer would be helpful
If direct contact is required please use demo@manx.net
I will be delighted to help
jsm

KeithMyers, 2019-03-20:
I looked at your sensors output and I have no clue why I thought the DIMM2 readings were 0°C.  See that they are the normal 216°C. unterminated values.  So I assume that the DIMM modules if they have internal temp sensors don't export via WMI protocol and thus are invisible to the asus-wmi-sensors driver.

As I stated earlier, any unterminated temp sensor input header has infinite resistance which scales to 216° C.  Only if you plug in a 10K thermistor sensor to one of the motherboard temp sensor input headers will you get a realistic and normal reading.

electrified, 2019-03-22:
Hi,

The DIMM.2 readings are for the headers on the M2 daughterboards rather than RAM. See top right in this picture. https://bit-tech.net/media/image/2017/8/6b49d5b7-30ef-42c3-bfb4-439279f31748.jpg

Are you using the daughterboards? Do you have any sensors connected to them?

As Keith says, 216 would be the usual value expected when no sensor present, 0 would be different, but the driver just reports the BIOS provided values.

If you have nothing connected I would ignore those readings in the sensors config file. The example file shows how to do that: https://github.com/lm-sensors/lm-sensors/blob/master/etc/sensors.conf.eg

KeithMyers, 2019-03-22:
Thanks for explaining the definition and locations of the DIMM.2 reading Ed.  I had never  seen a picture of the daughterboard adapter in detail. Ignore the readings as Ed says.

electrified, 2019-03-31:
I'm going to close this as I don't think there is anything to do here the )deg readings are an oddity, but ignore any not in use sensors via config.

closed 2019-03-31 by electrified