electrified/asus-wmi-sensors #11: ASUS Prime B450-Plus

guedressel, 2019-03-22:
Please add support for ASUS Prime B450-Plus

````
# dmidecode 3.1
Getting SMBIOS data from sysfs.
SMBIOS 3.1.1 present.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: PRIME B450-PLUS
        Version: Rev X.0x
        Serial Number: 181140080802612
        Asset Tag: Default string
        Features:
                Board is a hosting board
                Board is replaceable
        Location In Chassis: Default string
        Chassis Handle: 0x0003
        Type: Motherboard
        Contained Object Handles: 0

Handle 0x0020, DMI type 10, 6 bytes
On Board Device Information
        Type: Video
        Status: Enabled
        Description:    To Be Filled By O.E.M.

Handle 0x0025, DMI type 41, 11 bytes
Onboard Device
        Reference Designation:  Onboard IGD
        Type: Video
        Status: Enabled
        Type Instance: 1
        Bus Address: 0000:00:02.0
````

guedressel, 2019-03-22:
Maybe this gives some more information: [siv64 sensor debug](https://github.com/electrified/asus-wmi-sensors/files/2996299/ASUS.Prime.B450-Plus.-.siv.sensor.debug.txt)

electrified, 2019-03-22:
Hi there,

If the ASUS Prime B450-Plus is like the Strix B450-F, the WMI version implemented in the BIOS is too old.

Can you get WMI sensors output in SIV or HWiNFO? In HWiNFO it specifically says if it is using WMI, in SIV there are ASUS WMI screens available if it is using it. See https://github.com/electrified/asus-wmi-sensors/issues/6#issuecomment-458482648 for some screenshots of when it is using WMI.

Your SIV output above is being read directly from the IT8665E sensor chip rather than using the WMI output.

guedressel, 2019-03-22:
[image]

No WMI available then?

electrified, 2019-03-22:
That looks like the case. I think the only option is the unmaintained out
of tree it87 driver. https://github.com/a1wong/it87. This may not work on
recent kernels.

It's a shame Asus uses ITE sensor chips, the nuvoton chips used by Gigabyte
and Asrock have publicly available data sheets and are therefore well
supported in mainline Linux.

guedressel, 2019-03-22:
Yep - I can confirm that the linked it87 driver does work on my Linux 5.0.

KeithMyers, 2019-03-22:
I was very happy when the asus-wmi-sensor driver appeared as the it87 driver was mostly useless on my ASUS C7H hosts.  I had exactly two fan headers report and maybe one temp.  Pretty much useless.  It worked pretty well on my ASUS X370 Prime Pro motherboards though.

electrified, 2019-03-31:
I've updated the README to indicate this is currently an unsupported board :( Hopefully ASUS update the WMI interface in a BIOS update.

closed 2019-03-31 by electrified

thehans, 2019-07-26:
Hello, I have the same board.  

From this post, it looks like BIOS 0809 was used to report this info:
> Maybe this gives some more information: [siv64 sensor debug](https://github.com/electrified/asus-wmi-sensors/files/2996299/ASUS.Prime.B450-Plus.-.siv.sensor.debug.txt)

The most recent update is BIOS 1608, is it possible that WMI was updated between these versions?  (I have 1607 actually at the moment).  Is there anything else I can check from Linux to verify this?  I don't have Windows on this machine.

[code]
$ dmidecode -t baseboard
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.1.1 present.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: PRIME B450-PLUS
        Version: Rev X.0x
        Serial Number: [redacted]
        Asset Tag: Default string
        Features:
                Board is a hosting board
                Board is replaceable
        Location In Chassis: Default string
        Chassis Handle: 0x0003
        Type: Motherboard
        Contained Object Handles: 0

Handle 0x0020, DMI type 10, 6 bytes
On Board Device Information
        Type: Video
        Status: Enabled
        Description:    To Be Filled By O.E.M.

Handle 0x0025, DMI type 41, 11 bytes
Onboard Device
        Reference Designation:  Onboard IGD
        Type: Video
        Status: Enabled
        Type Instance: 1
        Bus Address: 0000:00:02.0
[/code]

electrified, 2019-07-29:
@thehans 

Support has not been added.

The easiest way to check is to check for the presence of the ACPI methods that the WMI interface uses.

This can be checked by downloading the BIOS image, opening it in UEFITool and searching for the text "SENX" which is one of the ACPI methods used by the WMI sensors interface. (Disable Unicode in the search dialogue)  It should be present within the DSDT.

Here are some images from the CH7, where you can see the search has found the string:

[image]

[image]

Whereas on the B450-Plus 1608 BIOS, SENX isn't present.

thehans, 2019-07-29:
Bummer, thanks for checking.