electrified/asus-wmi-sensors #14: make on Debian fails

iFA88, 2019-04-19:
Hi!
During the make a header file missing:

```
  CC [M]  /usr/src/asus-wmi-sensors-83a48b0/asus-wmi-sensors.o
/usr/src/asus-wmi-sensors-83a48b0/asus-wmi-sensors.c:18:23: fatal error: linux/wmi.h: No such file or directory
 #include <linux/wmi.h>
                       ^
compilation terminated.
/usr/src/linux-headers-4.9.0-8-common/scripts/Makefile.build:315: recipe for target '/usr/src/asus-wmi-sensors-83a48b0/asus-wmi-sensors.o' failed
make[4]: *** [/usr/src/asus-wmi-sensors-83a48b0/asus-wmi-sensors.o] Error 1
/usr/src/linux-headers-4.9.0-8-common/Makefile:1546: recipe for target '_module_/usr/src/asus-wmi-sensors-83a48b0' failed
make[3]: *** [_module_/usr/src/asus-wmi-sensors-83a48b0] Error 2
Makefile:152: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:72: recipe for target 'modules' failed
make: *** [modules] Error 2
```

I don't find any packages which satisfies this.

pixel1001, 2019-04-21:
I got the same issue,
but after upgrade the kernel, I could compiled it.

I had linux-image-4.9.0-8-amd64
And installed the backports linux-image-4.19.0-0.bpo.4-amd64

I don't know what is the difference, but seems that in newer kernel version that library is present.
Hope it helps.

iFA88, 2019-04-21:
Thank you for the tip!
I have downloaded `linux-image-4.19.0-0.bpo.4-amd64` with `linux-headers-4.9.0-7-amd64` but did not help. :(

iFA88, 2019-04-21:
I have modified `Makefile`:
```
#ifndef TARGET
#TARGET         := $(shell uname -r)
#endif
# Or specific version
TARGET          := 4.19.0-0.bpo.4-amd64
```

And now i have compiled :)

closed 2019-04-21 by iFA88

electrified, 2019-04-21:
Glad you got it sorted.

As you have found kernel version >= 4.12 is required to use the module.