electrified/asus-wmi-sensors #15: Request: Asus Zenith Extreme Alpha Motherboard

thagabe, 2019-04-21:
I have recently updated my motherboard to the new "revision" of the Zenith Extreme. 

dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.0.0 present.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
Manufacturer: ASUSTeK COMPUTER INC.
Product Name: ROG ZENITH EXTREME ALPHA
Version: Rev 1.xx
Serial Number: xxxxxxxxxxxxxxxx
Asset Tag: Default string
Features:
Board is a hosting board
Board is removable
Board is replaceable
Location In Chassis: Default string
Chassis Handle: 0x0003
Type: Motherboard
Contained Object Handles: 0

electrified, 2019-04-22:
Hi, I've pushed a new version including the DMI board name for the Alpha. Let me know if it works. 

If it doesn't please post the dmesg output from the module as this should help with debugging why.

thagabe, 2019-04-30:
It works! but outputs are weird, I blame asus happens on hwmon under windows too.

closed 2019-04-30 by thagabe