electrified/asus-wmi-sensors #17: Building on 5.0.10 failed

Jjeje007, 2019-05-02:
Try to build against linux 5.0.10 (-zen) on gentoo x64 failed :

`DKMS make.log for asus-wmi-sensors-ecf1322 for kernel 5.0.10-zen+ (x86_64)
jeu. mai  2 21:00:07 UTC 2019
  CC [M]  /var/lib/dkms/asus-wmi-sensors/ecf1322/build/asus-wmi-sensors.o
/var/lib/dkms/asus-wmi-sensors/ecf1322/build/asus-wmi-sensors.c: Dans la fonction « asus_wmi_call_method »:
/var/lib/dkms/asus-wmi-sensors/ecf1322/build/asus-wmi-sensors.c:134:11: error: déclaration implicite de la fonction « wmi_evaluate_method »; vouliez-vous utiliser « wmidev_evaluate_method » ? [-Werror=implicit-function-declaration]
  status = wmi_evaluate_method(ASUS_HW_GUID,
           ^~~~~~~~~~~~~~~~~~~
           wmidev_evaluate_method
/var/lib/dkms/asus-wmi-sensors/ecf1322/build/asus-wmi-sensors.c: Dans la fonction « asus_wmi_probe »:
/var/lib/dkms/asus-wmi-sensors/ecf1322/build/asus-wmi-sensors.c:612:7: error: déclaration implicite de la fonction « wmi_has_guid »; vouliez-vous utiliser « cpu_has_bug » ? [-Werror=implicit-function-declaration]
  if (!wmi_has_guid(ASUS_HW_GUID)) {
       ^~~~~~~~~~~~
       cpu_has_bug
cc1 : certains avertissements sont traités comme des erreurs
make[3]: *** [scripts/Makefile.build:283: /var/lib/dkms/asus-wmi-sensors/ecf1322/build/asus-wmi-sensors.o] Error 1
make[2]: *** [Makefile:1579: _module_/var/lib/dkms/asus-wmi-sensors/ecf1322/build] Error 2
make[1]: *** [Makefile:72: modules] Error 2
`

I build dkms for git source at the same time. Gcc-8.3.0 lm_sensors-3.5.0. Everything up to date (running ~amd64 aka testing).

If you need more info just ask.

Thx

electrified, 2019-05-03:
Hi, I'm on Arch running 5.0.10 and for me it builds correctly:

```
ed@zoomer ~/dev/github/asus-wmi-sensors (git)-[master] % git pull
Already up to date.
ed@zoomer ~/dev/github/asus-wmi-sensors (git)-[master] % git status
On branch master
Your branch is up to date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)

        .vscode/

nothing added to commit but untracked files present (use "git add" to track)
ed@zoomer ~/dev/github/asus-wmi-sensors (git)-[master] % make clean
ed@zoomer ~/dev/github/asus-wmi-sensors (git)-[master] % make
  CC [M]  /home/ed/dev/github/asus-wmi-sensors/asus-wmi-sensors.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/ed/dev/github/asus-wmi-sensors/asus-wmi-sensors.mod.o
  LD [M]  /home/ed/dev/github/asus-wmi-sensors/asus-wmi-sensors.ko
ed@zoomer ~/dev/github/asus-wmi-sensors (git)-[master] % uname -a
Linux zoomer 5.0.10-arch1-1-ARCH #1 SMP PREEMPT Sat Apr 27 20:06:45 UTC 2019 x86_64 GNU/Linux
2 ed@zoomer ~/dev/github/asus-wmi-sensors (git)-[master] % sudo make dkms                                                                                          :(
[sudo] password for ed: 

Creating symlink /var/lib/dkms/asus-wmi-sensors/ecf1322/source ->
                 /usr/src/asus-wmi-sensors-ecf1322

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j16 KERNELRELEASE=5.0.10-arch1-1-ARCH TARGET=5.0.10-arch1-1-ARCH...
cleaning build area...
Kernel cleanup unnecessary for this kernel.  Skipping...

DKMS: build completed.

asus-wmi-sensors:
Running module version sanity check.
 - Original module
   - An original module was already stored during a previous install
 - Installation
   - Installing to /usr/lib/modules/5.0.10-arch1-1-ARCH/kernel/drivers/hwmon/asus-wmi-sensors/

depmod....

DKMS: install completed.
sudo make dkms  6.63s user 0.96s system 52% cpu 14.487 total
ed@zoomer ~/dev/github/asus-wmi-sensors (git)-[master] % 
```

Jjeje007, 2019-05-04:
Hello,

I investigate a little bit. I was missing the ACPI_WMI and WMI_BMOF kernel module (found under -> Device Drivers -> X86 Platform Specific Device Drivers). So your module depends on this... should be good to point it out in the README. 

For now, every thing is working good, thanks for your time and good job !

Jérôme.

electrified, 2019-05-04:
Thanks, I will update the readme to mention this!

electrified, 2019-05-06:
I've updated the readme to say these modules are required. Thanks.

closed 2019-05-06 by electrified