electrified/asus-wmi-sensors #19: Prime X399-A (Bios: 1002)

dl7tny, 2019-06-06:
Just want to stop by and tell you, that the Prime X399-A (Bios: 1002) Board now has WMI Version 2 support and after "whitelisting" the Board in your driver it seems to work without problems. I have made the following changes to your code:

```
--- asus-wmi-sensors/asus-wmi-sensors.c	2019-06-06 13:47:45.517306450 +0200
+++ asus-wmi-sensors.dl7tny/asus-wmi-sensors.c	2019-06-06 13:31:11.586153343 +0200
@@ -32,6 +32,7 @@
 #define ZENITH_EXTREME "ROG ZENITH EXTREME"
 #define ZENITH_EXTREME_ALPHA "ROG ZENITH EXTREME ALPHA"
 #define PRIME_X470_PRO "PRIME X470-PRO"
+#define PRIME_X399_A "PRIME X399-A"

 #define METHODID_SENSOR_GET_VALUE     		0x52574543
 #define METHODID_SENSOR_UPDATE_BUFFER     	0x51574543
@@ -545,6 +546,7 @@
 			strcmp(board_name, CROSSHAIR_6_EXTREME) == 0 ||
 			strcmp(board_name, ZENITH_EXTREME) == 0 ||
 			strcmp(board_name, ZENITH_EXTREME_ALPHA) == 0 ||
+			strcmp(board_name, PRIME_X399_A) == 0 ||
 			strcmp(board_name, PRIME_X470_PRO) == 0))) {

 			pr_info("asuswmisensors: Supported board");
```

And the output looks very promising...

```
asuswmisensors-isa-0000
Adapter: ISA adapter
CPU Core Voltage:         +0.83 V
CPU SOC Voltage:          +0.83 V
DRAM AB Voltage:          +1.18 V
DRAM CD Voltage:          +1.18 V
1.8V PLL Voltage:         +1.81 V
+12V Voltage:            +11.90 V
+5V Voltage:              +5.01 V
3VSB Voltage:             +3.31 V
VBAT Voltage:             +3.10 V
AVCC3 Voltage:            +3.31 V
SB 1.05V Voltage:         +1.06 V
CPU Core Voltage:         +0.84 V
CPU SOC Voltage:          +0.83 V
DRAM AB Voltage:          +1.20 V
DRAM CD Voltage:          +1.19 V
CPU Fan:                 4192 RPM
Chassis Fan 1:           2812 RPM
Chassis Fan 2:           2848 RPM
Chassis Fan 3:              0 RPM
Water Pump 1:               0 RPM
CPU OPT:                 2786 RPM
EXT Fan 1:                  0 RPM
EXT Fan 2:                  0 RPM
EXT Fan 3:                  0 RPM
Cover Fan:                  0 RPM
CPU Temperature:          +34.0°C
CPU Socket Temperature:   +34.0°C
Motherboard Temperature:  +30.0°C
Chipset Temperature:      +41.0°C
Tsensor Temperature:     +216.0°C
CPU VRM Temperature:      +32.0°C
EXT Tsensor 1:             +0.0°C
EXT Tsensor 2:             +0.0°C
EXT Tsensor 3:             +0.0°C
CPU VRM Output Current:   +0.00 A
```

KeithMyers, 2019-06-06:
Thanks for the notification.  One more platform that is compatible.

electrified, 2019-06-07:
Thanks for this, I will incorporate support into the driver.

electrified, 2019-06-11:
Support for PRIME X399-A has been added.

closed 2019-06-11 by electrified