electrified/asus-wmi-sensors #20: Gentoo Linux

NightDragon1, 2019-06-16:
Hi!

I just created an ebuild for gentoo linux, accessing your repository. You may add that info to your readme.
To install the driver using Gentoo the following commands can be used:

1. Install layman (repository manager)
emerge layman
2. Add the repository to gentoo
layman -o https://github.com/NightDragon1/nightdragon_layman/master/layman.xml -f -a nightdragon_layman
3. Install the driver:
 emerge -av asus-wmi-sensors
4. load module:
modprobe asus-wmi-sensors

The Sensors should now show up using the "sensors" command from the lm_sensors package

BR,
Robert

gyakovlev, 2019-06-18:
hey, I also have an ebuild here https://github.com/gyakovlev/gentoo-overlay/tree/master/sys-kernel/asus-wmi-sensors
feel free to simplify your ebuild based on mine.

I'm going to test ebuild a bit and probably will move it to main gentoo tree if the module causes no trouble.
works fine on PRIME X399-A so far
```
asuswmisensors-isa-0000
Adapter: ISA adapter
CPU Core Voltage:         +0.85 V  
CPU SOC Voltage:          +1.11 V  
DRAM AB Voltage:          +1.40 V  
DRAM CD Voltage:          +1.42 V  
1.8V PLL Voltage:         +1.88 V  
+12V Voltage:            +12.23 V  
+5V Voltage:              +5.04 V  
3VSB Voltage:             +3.33 V  
VBAT Voltage:             +3.12 V  
AVCC3 Voltage:            +3.36 V  
SB 1.05V Voltage:         +1.05 V  
CPU Core Voltage:         +0.78 V  
CPU SOC Voltage:          +1.04 V  
DRAM AB Voltage:          +1.35 V  
DRAM CD Voltage:          +1.35 V  
CPU Fan:                  734 RPM
Chassis Fan 1:            621 RPM
Chassis Fan 2:            647 RPM
Chassis Fan 3:            783 RPM
Water Pump 1:               0 RPM
CPU OPT:                  740 RPM
EXT Fan 1:                  0 RPM
EXT Fan 2:                  0 RPM
EXT Fan 3:                  0 RPM
Cover Fan:                  0 RPM
CPU Temperature:          +32.0°C  
CPU Socket Temperature:   +33.0°C  
Motherboard Temperature:  +26.0°C  
Chipset Temperature:      +43.0°C  
Tsensor Temperature:      +34.0°C  
CPU VRM Temperature:      +32.0°C  
EXT Tsensor 1:             +0.0°C  
EXT Tsensor 2:             +0.0°C  
EXT Tsensor 3:             +0.0°C  
CPU VRM Output Current:   +0.00 A  
```

stefson, 2019-06-28:
Do you need to have uefi boot enabled if you're going to use the wmi sensors?

KeithMyers, 2019-06-28:
If you are referring to Secure Boot, then no.

electrified, 2020-02-27:
I've had success with @gyakovlev's ebuild, and his overlay is well maintained, so have updated the readme to include details. Many thanks to both of you for your packaging efforts.

closed 2020-02-27 by electrified