electrified/asus-wmi-sensors #21: Getting an Error: Could not insert 'asus_wmi_sensors' : No Such Device

terroreek, 2019-07-04:
I am trying to use this with the Asus X399-e Strix motherboard with Siduction (Debian Unstable).

I know its not on the list of supported Motherboards but I thought I would try it, it appears to work fine until I try to modeprobe the module.

I am running Kernel 5.1.15, here is the current output from sensors.
`asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:    +1.00 V  
SVI2_SoC:     +0.01 V  
Tdie:         +38.6°C  (high = +70.0°C)
Tctl:         +65.6°C  
SVI2_P_Core:  58.20 W  
SVI2_P_SoC:    0.00 W  
SVI2_C_Core: +58.20 A  
SVI2_C_SoC:   +0.00 A  

zenpower-pci-00cb
Adapter: PCI adapter
SVI2_Core:    +0.83 V  
SVI2_SoC:     +0.01 V  
Tdie:         +37.4°C  (high = +70.0°C)
Tctl:         +64.4°C  
SVI2_P_Core:   5.19 W  
SVI2_P_SoC:    0.00 W  
SVI2_C_Core:  +7.27 A  
SVI2_C_SoC:   +0.00 A  

`

I can provide the debug logs from HWMon if that helps.

electrified, 2019-07-04:
Could you provide the `dmesg` output from when you insert the module?

And also a screenshot of HWiNFO to see if that is using the WMI interface.

Thanks.

terroreek, 2019-07-04:
This is what I see in dmesg 

`[10656.640029] asuswmisensors: Vendor: ASUSTeK COMPUTER INC. Board: ROG STRIX X399-E GAMING BIOS version: 1002 WMI version: 2`

Here are the screenshots I believe you are looking for.  Attached is also the debug logs, running in sensors only.  

[image]

[image]

[HWiNFO64.zip](https://github.com/electrified/asus-wmi-sensors/files/3360191/HWiNFO64.zip)

electrified, 2019-07-05:
Thanks! From that output I can add support. I'll try and do it over the weekend.

KeithMyers, 2019-07-05:
Have a question for Ed.  I'm seeing new AGESA Combo BIOS' for the new X570 motherboards being released.  I also saw news that ASUS has dropped the ITE 8665E SIO chipset in favor of Nuvoton SIO chipsets.  I also see HwInfo64 screenshots of these new Combo BIOS' that show a WMI interface.

Question: Will the asus-wmi-sensor driver seamlessly handle the changeover from ITE to Nuvoton SIO chipset simply because the WMI interface does not care about the low-level SIO chip?

electrified, 2019-07-05:
@KeithMyers Hey Keith, unless Asus have radically changed the way the WMI interface works, I expect it should "just work" (may have to add add BIOS strings depending on what they've done with the version number).

Alternatively though, the Nuvoton NCT6798D which is what they are using has mainline kernel support, so it should be possible to access the sensors via that instead of the WMI interface.

KeithMyers, 2019-07-05:
Hi Ed, yes it wouldn't be the end of the world to have to use the mainline Nuvoton drivers.  I use the NCT6775 driver for my Asrock X399 board and it works.  Doesn't offer the plethora of data that the asus-wmi-sensor driver does, but is sufficient for the basic parameters.

At some point I will be the guinea pig for the new combo BIOS and one of the Ryzen 3000 cpus in my ASUS C7H boards.  So far I have just stuck to the old 1002 BIOS which is the first good working version of the WMI BIOS.  I haven't bothered to try all the dozens of beta BIOS' that all the overclockers have been beta testing.  I will wait for the dust to settle first.  To give you an idea of what the current latest BIOS is doing . . . . . install it and no mouse works in the BIOS anymore!  Great quality control ASUS.

electrified, 2019-07-06:
@terroreek I've added support now. Can you give it a go and see if it works please?

Thanks.

terroreek, 2019-07-06:
Its working!!

`asuswmisensors-isa-0000
Adapter: ISA adapter
CPU Core Voltage:         +1.20 V  
CPU SOC Voltage:          +1.02 V  
DRAM AB Voltage:          +1.33 V  
DRAM CD Voltage:          +1.33 V  
1.8V PLL Voltage:         +1.81 V  
+12V Voltage:            +11.84 V  
+5V Voltage:              +4.93 V  
3VSB Voltage:             +3.29 V  
VBAT Voltage:             +3.10 V  
AVCC3 Voltage:            +3.29 V  
SB 1.05V Voltage:         +1.04 V  
CPU Core Voltage:         +0.84 V  
CPU SOC Voltage:          +1.00 V  
DRAM AB Voltage:          +1.35 V  
DRAM CD Voltage:          +1.35 V  
CPU Fan:                 2436 RPM
Chassis Fan 1:           1056 RPM
Chassis Fan 2:              0 RPM
Chassis Fan 3:              0 RPM
Water Pump 1:               0 RPM
CPU OPT:                    0 RPM
EXT Fan 1:                  0 RPM
EXT Fan 2:                  0 RPM
EXT Fan 3:                  0 RPM
Cover Fan:                  0 RPM
CPU Temperature:          +35.0°C  
CPU Socket Temperature:   +32.0°C  
Motherboard Temperature:  +30.0°C  
Chipset Temperature:      +49.0°C  
Tsensor Temperature:      +36.0°C  
CPU VRM Temperature:      +35.0°C  
EXT Tsensor 1:             +0.0°C  
EXT Tsensor 2:             +0.0°C  
EXT Tsensor 3:             +0.0°C  
CPU VRM Output Current:   +0.00 A`

terroreek, 2019-07-06:
since everything appears to be working I am going to close this issue.

closed 2019-07-06 by terroreek