electrified/asus-wmi-sensors #22: Can you add ASUS Pro WS X570-ACE motherboard ?

pducharme, 2019-07-13:
Hi, I'm using the new Ryzen 9 3900x on a "workstation-class" motherboard from ASUS, the ASUS Pro WS X570-ACE motherboard.  This motherboard has Asus Control utility with the sensors reading, so it's probably possible to make a wmi support for it ?   That would be really cool!

electrified, 2019-07-14:
Could you provide the dmesg output from when you insert the module? This should be logged even if the driver doesn't yet support your motherboard.

e.g 
`[    5.177760] asuswmisensors: Vendor: ASUSTeK COMPUTER INC. Board: ROG CROSSHAIR VII HERO (WI-FI) BIOS version: 2406 WMI version: 2`

Or alternatively the output from `sudo dmidecode -t baseboard`

Additionally a screenshot of the HWiNFO sensors page in Windows to verify if that is using the WMI interface, and the sensor values returned.

Thanks.

pducharme, 2019-07-14:
Ok, here is the output of HW Sensors page.
[image]
[image]
[image]
[image]

pducharme, 2019-07-14:
Here is the output of dmidecode  : 

`Linux 4.19.56-Unraid.
root@Tower:~# sudo dmidecode -t baseboard
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: Pro WS X570-ACE
        Version: Rev X.0x
        Serial Number: 190552238800388
        Asset Tag: Default string
        Features:
                Board is a hosting board
                Board is replaceable
        Location In Chassis: Default string
        Chassis Handle: 0x0003
        Type: Motherboard
        Contained Object Handles: 0

Handle 0x0027, DMI type 10, 6 bytes
On Board Device Information
        Type: Video
        Status: Enabled
        Description:    To Be Filled By O.E.M.

Handle 0x002D, DMI type 41, 11 bytes
Onboard Device
        Reference Designation:  Onboard IGD
        Type: Video
        Status: Enabled
        Type Instance: 1
        Bus Address: 0000:00:02.0

root@Tower:~# `

electrified, 2019-07-15:
Thanks for this.

Interesting that HWiNFO is using the Nuvoton hardware monitoring IC directly rather than the WMI interface. 

I will add the motherboard to the list of devices supported and we will see what happens!

electrified, 2019-07-15:
OK, I've added the motherboard identification string.

Let me know if it works or not, and what output you get from the driver in `dmesg`

If it doesn't work, I may have to send you some PowerShell scripts to interrogate the WMI interface in windows.

Thanks.

pducharme, 2019-07-15:
i'm not sure I have all what is needed to build on UnRaid 6.7.2, it's a minimal installation of storage server running out of a USB key.

electrified, 2019-07-15:
Hi, I haven't got any experience with unRaid - if you can get the usual kernel development toolchain installed e.g. gcc, make, kernel sources etc you should be able to use the Makefile to build the module.

bvanalderweireldt, 2019-07-15:
What about the ASUS Prime X570-Pro ?

>>>
# dmidecode 3.1
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.
# SMBIOS implementations newer than version 3.1.1 are not
# fully supported by this version of dmidecode.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: PRIME X570-PRO
	Version: Rev X.0x
	Serial Number: 190652525900800
	Asset Tag: Default string
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Default string
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x002D, DMI type 10, 6 bytes
On Board Device Information
	Type: Video
	Status: Enabled
	Description:    To Be Filled By O.E.M.

Handle 0x0033, DMI type 41, 11 bytes
Onboard Device
	Reference Designation:  Onboard IGD
	Type: Video
	Status: Enabled
	Type Instance: 1
	Bus Address: 0000:00:02.0

electrified, 2019-07-15:
@bvanalderweireldt Hey. I've added your motherboards identifier, but this is completely experimental.

Give it a go and see if it works and what the `dmesg` output reports.

electrified, 2019-07-17:
I've downloaded a few of the Asus X570 BIOS images (Crosshair VIII, PRO WS) and had a look at the ACPI DSDT tables.

To me it looks like they've removed the WMI sensors interface, so I don't think it will be possible to support these boards.

I'm going to ask other monitoring software authors to confirm, but the mainline Nuvoton driver is probably the best bet.

electrified, 2019-07-17:
OK, I've had it confirmed that the WMI interface is no more on X570. Sorry! I've removed the code I added.

The `nct6775` driver, in kernels > 5.0 should support the NCT6797D present on your boards.

Tagging @KeithMyers as I know he was interested in if support was possible.

closed 2019-07-17 by electrified

KeithMyers, 2019-07-17:
I have a hunch there just isn't room in the BIOS for the WMI interface.  The current combo BIOS' are double sized already due needing to support all the older cpus along with the new cpus.  Maybe later in BIOS development when the code can be simplified, they might add the WMI interface to the X570 motherboards.  Or they could be simply happy with the Nuvoton SIO chip support.  ASUS does not appear to have scores of crack BIOS engineers releasing daily BIOS updates.  Not even weekly.  And rarely monthly.  I will be having to update my BIOS next week to support the 3900X I'm receiving.  I have been perfectly happy with the long stable 1002 BIOS for my 2700X and C7H board.  Fun times ahead I'm sure.

KeithMyers, 2019-07-18:
Thanks for the info Ed.  Shame that they have issues with the WMI interface on Matisse.  The WMI interface was already in use in Linux before the ITE SIO chip fiasco.  And still is.  Still see Elmor participating in the OCN forums on C7H boards offering help and suggestions.  Even when he doesn't work for ASUS anymore.

Would appreciate a dump of sensors -u with the NCT6775 driver in use on the 570-ACE board.  I am considering that one in the future.

techie2000, 2021-01-26:
@KeithMyers I'm not sure about the NCT6775 driver, but I am running that board and the result of `sensors -u` is 

> k10temp-pci-00c3
> Adapter: PCI adapter
> Vcore:
>   in0_input: 1.550
> Vsoc:
>   in1_input: 1.375
> CPU Temp:
>   temp1_input: 56.375
> Tdie:
>   temp2_input: 56.375
> Tccd1:
>   temp3_input: 57.500
> Tccd2:
>   temp4_input: 48.000
> Icore:
>   curr1_input: 0.000
> Isoc:
>   curr2_input: 14.260
> 
> nvme-pci-0100
> Adapter: PCI adapter
> Composite:
>   temp1_input: 40.850
>   temp1_max: 84.850
>   temp1_min: -273.150
>   temp1_crit: 84.850
>   temp1_alarm: 0.000
> Sensor 1:
>   temp2_input: 40.850
>   temp2_max: 65261.850
>   temp2_min: -273.150
> Sensor 2:
>   temp3_input: 40.850
>   temp3_max: 65261.850
>   temp3_min: -273.150

KeithMyers, 2021-01-27:
The NCT6775 driver in Linux has very limited output.  Basically useless.  Thankfully the zenpower driver reports the temps, powers and voltages of Zen 2 cpus for Ryzen, Threadripper and Epyc. The zenmonitor GUI is very nice also.
[https://github.com/ocerman](url)

hanetzer, 2021-09-17:
> Hi, I'm using the new Ryzen 9 3900x on a "workstation-class" motherboard from ASUS, the ASUS Pro WS X570-ACE motherboard. This motherboard has Asus Control utility with the sensors reading, so it's probably possible to make a wmi support for it ? That would be really cool!

@pducharme I hope I'm not out of line here, but I'm trying to do some research on the board you have for the purpose of
porting either coreboot or oreboot to it, and a proper bmc-like software on the rtl8117 management interface. If you're willing
(involves running various hardware probing commands under linux) I'd greatly appreciate it. You can find me under this nick
on the libera irc network, or shoot me an email, same name, at protonmail.

pducharme, 2021-09-17:
Hi, I’m sorry, I don’t have this Motherboard anymore.  Good luck and I hope you’ll be successful! That would be great to have a good bmc on this rtl8117 !

techie2000, 2021-09-20:
Hi @hanetzer 
I'm running the ASUS Pro WS X570-ACE board with the second to latest bios. I'll happily run you any linux cli probes. Note I'm running "Slackware 14.2 x86_64 (post 14.2 -current)" if that has an influence on the offer :)

hanetzer, 2021-10-01:
> Hi @hanetzer I'm running the ASUS Pro WS X570-ACE board with the second to latest bios. I'll happily run you any linux cli probes. Note I'm running "Slackware 14.2 x86_64 (post 14.2 -current)" if that has an influence on the offer :)

distro shouldn't matter much. Like I said to the original poster, find me on libera or email me to arrange some form of real time communication.

hanetzer, 2021-10-10:
US/Central, various gentoo channels and others. You can use the command `/whois hanetzer` to get a full list