electrified/asus-wmi-sensors #24: Enable ROG STRIX B450-I GAMING

michael42, 2019-07-16:
Hi there,

I think this module supports the Asus ROG STRIX B450-I GAMING (with the current Firmware 2406).

dmidecode:
```
Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: ROG STRIX B450-I GAMING
        Version: Rev 1.xx
        Serial Number: 180630735500051
        Asset Tag: Default string
        Features:
                Board is a hosting board
                Board is replaceable
        Location In Chassis: Default string
        Chassis Handle: 0x0003
        Type: Motherboard
        Contained Object Handles: 0
```

I patched the module to enable it and got the following `sensors` output, which seems plausible for my 3700X:
```
asuswmisensors-isa-0000
Adapter: ISA adapter
CPU Core Voltage:         +0.88 V  
VPP MEM Voltage:          +2.50 V  
+12V Voltage:             +9.87 V  
+5V Voltage:              +5.01 V  
3VSB Voltage:             +3.33 V  
VBAT Voltage:             +3.27 V  
AVCC3 Voltage:            +3.33 V  
SB 1.05V Voltage:         +1.07 V  
CPU Core Voltage:         +0.90 V  
CPU SOC Voltage:          +1.04 V  
CPU Fan:                 2242 RPM
Chassis Fan 1:              0 RPM
AIO Pump:                   0 RPM
CPU OPT:                    0 RPM
CPU Temperature:          +56.0°C  
CPU Socket Temperature:   +49.0°C  
Motherboard Temperature:  +54.0°C  
Chipset Temperature:      +56.0°C  
Tsensor 1 Temperature:   +216.0°C  
CPU VRM Temperature:      +47.0°C  
CPU VRM Output Current:   +2.00 A  
```

Thanks for the great work!

electrified, 2019-07-17:
Many thanks - I will add support for this board too

electrified, 2019-07-17:
Support added.

closed 2019-07-17 by electrified

super119, 2019-07-22:
Just the +12v voltage looks wrong.