electrified/asus-wmi-sensors #25: Board not supported, not capable of modprobe

pbeliveau, 2019-07-20:
I've installed the asus-wmi-sensors-dkms-git from the AUR. However, I get this error message when trying to load the module:
`modprobe: ERROR: could not insert 'asus_wmi_sensors': No such device`

This is the motherboard:
- Asus ROG STRIX X470-I

Here is the output of the dmesg:
`[    3.820114] usb 1-10: Manufacturer: AsusTek Computer Inc.
[    4.665407] asus_wmi_sensors: loading out-of-tree module taints kernel.
[    4.665417] asus_wmi_sensors: module verification failed: signature and/or required key missing - tainting kernel
[    4.665792] asuswmisensors: Vendor: ASUSTeK COMPUTER INC. Board: ROG STRIX X470-I GAMING BIOS version: 2406 WMI version: 2
[    4.665793] asuswmisensors: Unsupported board
[    4.858623] asus_wmi: ASUS WMI generic driver loaded
[    4.859769] asus_wmi: Initialization: 0x0
[    4.859791] asus_wmi: BIOS WMI version: 0.9
[    4.859843] asus_wmi: SFUN value: 0x0
[    4.860280] asus_wmi: Number of fans: 1
[    5.030254] hid-generic 0003:0B05:1872.0003: hiddev0,hidraw2: USB HID v1.11 Device [AsusTek Computer Inc. AURA LED Controller] on usb-0000:02:00.0-10/input0
[   48.912521] asuswmisensors: Vendor: ASUSTeK COMPUTER INC. Board: ROG STRIX X470-I GAMING BIOS version: 2406 WMI version: 2
`

This the uname output:
`5.2.1-arch1-1-ARCH`

Can support for the ROG STRIX X470-I GAMING be added?

electrified, 2019-07-20:
Hi, I've added the board identifier string. 

Let me know how it works.

pbeliveau, 2019-07-20:
Happy to report it fixes the issue.

Dmesg output
```
[    4.698888] asuswmisensors: Vendor: ASUSTeK COMPUTER INC. Board: ROG STRIX X470-I GAMING BIOS version: 2501 WMI version: 2
[    4.698889] asuswmisensors: Supported board
[    4.698889] asuswmisensors: ASUS WMI sensors driver loaded
```

Sensors output:
```
asuswmisensors-isa-0000          
Adapter: ISA adapter             
CPU Core Voltage:         +0.17 V
VPP MEM Voltage:          +2.47 V
+12V Voltage:             +9.76 V
+5V Voltage:              +4.96 V
3VSB Voltage:             +3.33 V
VBAT Voltage:             +3.21 V
AVCC3 Voltage:            +3.31 V
SB 1.05V Voltage:         +1.05 V
CPU Core Voltage:         +0.97 V
CPU SOC Voltage:          +1.08 V
CPU Fan:                 1520 RPM
Chassis Fan 1:              0 RPM
AIO Pump:                2115 RPM
CPU OPT:                    0 RPM
CPU Temperature:          +32.0°C
CPU Socket Temperature:   +33.0°C
Motherboard Temperature:  +33.0°C
Chipset Temperature:      +54.0°C
Tsensor 1 Temperature:   +216.0°C
CPU VRM Temperature:      +36.0°C
CPU VRM Output Current:   +3.00 A
```

Closing the issue.

closed 2019-07-20 by pbeliveau