electrified/asus-wmi-sensors #26: cant compile

vdub12, 2019-07-21:
getting the error "fatal error: linux/wmi.h: No such file or directory" when I try and compile. 

Do you know if this will work on the TUF B450M-Plus?

electrified, 2019-07-22:
Hi, what distribution are you using and what kernel version are you on?

Regarding the TUF B450M-Plus, I think it is unlikely that it has WMI support. 

If you use HWiNFO in Windows does it say it's using ASUS WMI or the ITE chip directly?

vdub12, 2019-07-22:
It's openSuse Leap 15. Kernel is 4.12.14

electrified, 2019-07-22:
The WMI interface is not present in 4.12.14, it was introduced in 14.13, hence the error you are getting

https://elixir.bootlin.com/linux/v4.12.14/source/include/linux/wmi.h

vs

https://elixir.bootlin.com/linux/v4.13/source/include/linux/wmi.h

vdub12, 2019-07-22:
Awe, makes sense. Considering the TUF boards probably don't support WMI
anyway I'll wait until I upgrade Leap and try again.

Next time I'm in Windows I'll check it out. It typically takes a few hours
to reboot because of system lockups due to the it87 driver. I normally have
to reboot about 15+ times because sensors crashes and the fans drop to
their lowest speed and I get hard lockups.

Thanks.

electrified, 2019-07-22:
OK. I will close this issue for now. If you do find the board uses WMI under Windows open a new ticket and I will add the board ID.

closed 2019-07-22 by electrified