electrified/asus-wmi-sensors #31: Module build error

Lexouden, 2019-07-24:
When trying to install the module on CentOS 7 it fails to build the module.

It tells me:
>unknown field ‘read_string’ specified in initializer .read_string = asus_wmi_hwmon_read_string,

Build log:
`DKMS make.log for asus-wmi-sensors-e547824 for kernel 3.10.0-957.21.3.el7.x86_64 (x86_64)
Wed Jul 24 11:02:42 CEST 2019
  CC [M]  /var/lib/dkms/asus-wmi-sensors/e547824/build/asus-wmi-sensors.o
/var/lib/dkms/asus-wmi-sensors/e547824/build/asus-wmi-sensors.c:399:2: error: unknown field ‘read_string’ specified in initializer
  .read_string = asus_wmi_hwmon_read_string,
  ^
/var/lib/dkms/asus-wmi-sensors/e547824/build/asus-wmi-sensors.c:399:2: warning: initialization from incompatible pointer type [enabled by default]
/var/lib/dkms/asus-wmi-sensors/e547824/build/asus-wmi-sensors.c:399:2: warning: (near initialization for ‘asus_wmi_hwmon_ops.write’) [enabled by default]
make[3]: *** [/var/lib/dkms/asus-wmi-sensors/e547824/build/asus-wmi-sensors.o] Error 1
make[2]: *** [_module_/var/lib/dkms/asus-wmi-sensors/e547824/build] Error 2
make[1]: *** [modules] Error 2`

### **Extra info:**
Using: _Asus ROG Zenith Extreme Alpha_ with a _AMD Ryzen 2990WX_
Had the module installed on Ubuntu 18.04 a couple days ago without issue.

electrified, 2019-07-24:
There is another issue open where a user has managed to get it to build on CentOS 7 with some changes: https://github.com/electrified/asus-wmi-sensors/issues/29#issue-470835979

Happy to accept pull requests to make it work (that don't compromise functionality on newer kernels)

electrified, 2019-07-25:
Closing this as we have the other CentOS 7 issue #29

closed 2019-07-25 by electrified