electrified/asus-wmi-sensors #34: PWM to control fans speed?

terencode, 2019-08-04:
This might not be something you can do with this module, but I was wondering if there was a way to control the different fans I have on my crosshair hero VI through `pwmconfig` for example, which currently reports:
`There are no usable PWM outputs.`

I tried using the out-of-tree it87 module, but despite showing some available PWM outputs, it couldn't find any link between writing to them and a change in fan speed.

NightDragon1, 2019-08-06:
Hi!

As i know the out of tree it87 module doesn't work properly...
You might use the patched version of groeck. as he put down the repo, you can find a clone under: https://github.com/xdarklight/hwmon-it87

In general it would be a cool idea if electrified could take over some code to provide the pwm controls.

BR,
Rob

terencode, 2019-08-06:
hey @NightDragon1 thanks, it's working well although like all it87 modules, I had to do `modprobe it87 ignore_resource_conflict`, I hope this doesn't have any bad side-effects...

I'll leave the issue open until @electrified tells us if he can do anything about it :).

electrified, 2019-08-06:
The IT87 accesses the hardware directly, whereas this driver goes via ASUS' WMI interface. Mixing the 2 access methods in one driver would be messy and I don't want to go down that route.

The other option is that there are other WMI methods in addition to the sensors ones, but they all have generic names `asio_hw_funXX`. It's plausible that these are used by AI Suite for setting fan speeds, voltages etc, but reverse engineering what these do is beyond my skills (attaching a debugger to the Asus drivers in Windows and observing if they are called when you change fan speed settings in AI Suite).

The ideal situation would be if ITE would open up the documentation for the IT8665E so the IT87 driver could be fixed, but others have tried and failed to get these...

terencode, 2019-08-06:
Ok, I see your point. Should the issue still be left open in case of any progress?

NightDragon1, 2019-08-06:
I suggest to close it. If someone one day will uncover / reverse engineer the function calls, a new issue could be opened.

terencode, 2019-08-06:
Alright, thanks.

closed 2019-08-06 by terencode