electrified/asus-wmi-sensors #35: Possible support for ROG Maximux XI?

intrusdave, 2019-08-07:
I was hoping to get support for this board added.  Both SIV and HWINFO fully support this board.

`# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.2.1 present.
# SMBIOS implementations newer than version 3.2.0 are not
# fully supported by this version of dmidecode.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: ROG MAXIMUS XI HERO (WI-FI)
	Version: Rev 1.xx
	Serial Number: 181037619600435
	Asset Tag: Default string
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Default string
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x0031, DMI type 10, 6 bytes
On Board Device Information
	Type: Video
	Status: Enabled
	Description:    To Be Filled By O.E.M.

Handle 0x0061, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: Onboard - Other
	Type: Other
	Status: Enabled
	Type Instance: 1
	Bus Address: 0000:00:00.0

Handle 0x0062, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: Onboard - Other
	Type: Other
	Status: Enabled
	Type Instance: 2
	Bus Address: 0000:00:14.0

Handle 0x0063, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: Onboard - Other
	Type: Other
	Status: Enabled
	Type Instance: 3
	Bus Address: 0000:00:14.2

Handle 0x0064, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: Onboard - Ethernet
	Type: Ethernet
	Status: Enabled
	Type Instance: 1
	Bus Address: 0000:00:14.3

Handle 0x0065, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: Onboard - Other
	Type: Other
	Status: Enabled
	Type Instance: 4
	Bus Address: 0000:00:16.0

Handle 0x0066, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: Onboard - SATA
	Type: SATA Controller
	Status: Enabled
	Type Instance: 1
	Bus Address: 0000:00:17.0

Handle 0x0067, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: Onboard - Other
	Type: Other
	Status: Enabled
	Type Instance: 5
	Bus Address: 0000:00:1f.0

Handle 0x0068, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: Onboard - Sound
	Type: Sound
	Status: Enabled
	Type Instance: 1
	Bus Address: 0000:00:1f.3

Handle 0x0069, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: Onboard - Other
	Type: Other
	Status: Enabled
	Type Instance: 6
	Bus Address: 0000:00:1f.4

Handle 0x006A, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: Onboard - Other
	Type: Other
	Status: Enabled
	Type Instance: 7
	Bus Address: 0000:00:1f.5

Handle 0x006B, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: Onboard - Ethernet
	Type: Ethernet
	Status: Enabled
	Type Instance: 2
	Bus Address: 0000:00:1f.6
`

electrified, 2019-08-07:
Hi,

I've had a look at the latest 1105 BIOS in uefitool, and it doesn't appear to have a WMI sensors interface. As far as I know the WMI sensors interface has only been included on Ryzen boards.

Does HWiNFO show ASUS WMI like in this https://user-images.githubusercontent.com/6266419/51900744-58106400-23b6-11e9-8d53-9ce80ab52384.jpg screenshot?

electrified, 2019-08-14:
Closing this as 99% sure this board doesn't have a WMI interface. Reopen if it does....

closed 2019-08-14 by electrified