electrified/asus-wmi-sensors #37: Temperature always stays at 49 degree Celsius.

laichiaheng, 2019-08-21:
When I test it with `mprime -t`, it still stays at 49 degree Celsius with the constant fan speed.
CPU: Ryzen 3700X
MB: Crosshair VII Hero

electrified, 2019-08-22:
Hi there,
Unfortunately it seems there are issues with the WMI implementation in recent BIOSes with ComboPI AGESA. The issues seem more prevalent when using Ryzen 3000 CPUs, I've not been able to recreate any of them with my 2700X/CH7.

The HWiNFO developer has contacted ASUS about the issues, so we will see if ASUS fixes the problems in a future BIOS release.

At this moment I don't know of any change I can make within the driver to mitigate the issue, sorry.

electrified, 2019-09-28:
Hi @laichiaheng 

Can you try now with the latest 2801 BIOS and the latest version of this driver?

I think with both the issue should now be fixed.

laichiaheng, 2019-09-28:
@electrified Yes, they haven't happened since I upgraded the BIOS to 2801, thanks.

closed 2019-09-28 by laichiaheng