electrified/asus-wmi-sensors #38: ERROR: could not insert 'asus_wmi_sensors': No such device

jack-46, 2019-09-10:
I have

MB:    ASUS Prime X470-Pro
BIOS: Ver. 4207, of 12/07/2018

Running Ubuntu 18.04 with all updates
kernel:    5.0.0-27-generic

I followed the step, outlined in    https://github.com/electrified/asus-wmi-sensors

The module builds fine, but:    modprobe asus-wmi-sensors
results in:     modprobe: ERROR: could not insert 'asus_wmi_sensors': No such device

Unfortunately, I do not have a Windows to check as suggested.

What can I do to get it to work?

Regards

Jack

electrified, 2019-09-10:
Your BIOS is too old and doesn't have WMI sensors, as per table
https://github.com/electrified/asus-wmi-sensors/blob/master/README.md#supported-hardware
update to BIOS 4602 or later and try again.

jack-46, 2019-09-10:
Am 10.09.19 um 14:01 schrieb Ed:
> Your BIOS is too old and doesn't have WMI sensors, as per table
> https://github.com/electrified/asus-wmi-sensors/blob/master/README.md#supported-hardware
> update to BIOS 4602 or later and try again.

Thanks.
I'll upgrade (but probably not before the end of the month) and report then

Regards

Jack

jack-46, 2019-09-11:
Am 10.09.19 um 14:01 schrieb Ed:
> Your BIOS is too old and doesn't have WMI sensors, as per table
> https://github.com/electrified/asus-wmi-sensors/blob/master/README.md#supported-hardware
> update to BIOS 4602 or later and try again.

I update to version 5216 and (of course!) it works now.

As I wrote down the old version I mistakenly wrote 4702 (>4602) rather 
than 4207.
My eyes are bad:'(

Thanks for answering and making this available.

Regards

Jack

electrified, 2019-09-11:
No problem, glad you got it working.

closed 2019-09-11 by electrified