electrified/asus-wmi-sensors #40: Asus TUF Z370 Plus-Gaming

Young-3, 2019-10-22:
my system is unraid ，base on Slackware Linux distribution. I use ‘sensors’ according to the following:

```
root@Tower:~# sensors
acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

coretemp-isa-0000
Adapter: ISA adapter
CPU Temp:     +35.0°C  (high = +82.0°C, crit = +100.0°C)
Core 0:       +33.0°C  (high = +82.0°C, crit = +100.0°C)
Core 1:       +35.0°C  (high = +82.0°C, crit = +100.0°C)
Core 2:       +33.0°C  (high = +82.0°C, crit = +100.0°C)
Core 3:       +33.0°C  (high = +82.0°C, crit = +100.0°C)
Core 4:       +33.0°C  (high = +82.0°C, crit = +100.0°C)
Core 5:       +33.0°C  (high = +82.0°C, crit = +100.0°C)
```
BIOS 2401

only coretemp，but I need motherboard ，fans and others.
I'm looking for some solutions. Can I use your method? I'm not familiar with it.

KeithMyers, 2019-10-22:
No, I don't believe so.  This driver is for BIOS' that have a WMI interface.  Your board does not export sensors through a WMI interface it appears. Nothing mentioned on the ASUS BIOS page for the board.

You need to find out which SIO sensor your board uses through the normal sensors-detect probe and find the driver for that SIO chip.  Or visually find the chip on the motherboard and read its identifier.

Young-3, 2019-10-22:
**This is lm_sensors version:**
```
root@Tower:~# sensors -v
sensors version 3.5.0 with libsensors version 3.5.0
```

**This is my detection result:**
```
root@Tower:~# sensors-detect
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# Board: ASUSTeK COMPUTER INC. TUF Z370-PLUS GAMING
# Kernel: 5.3.6-Unraid x86_64
# Processor: Intel(R) Core(TM) i5-8400 CPU @ 2.80GHz (6/158/10)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no):
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no):
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd121
    (logical device B has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no):
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no):
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no):
Found unknown SMBus adapter 8086:a2a3 at 0000:00:1f.4.
Sorry, no supported PCI bus adapters found.
Module i2c-dev loaded successfully.

Next adapter: SMBus I801 adapter at f040 (i2c-0)
Do you want to scan it? (YES/no/selectively):
Client found at address 0x51
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No
Client found at address 0x53
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No

Next adapter: i915 gmbus dpc (i2c-1)
Do you want to scan it? (yes/NO/selectively):

Next adapter: i915 gmbus dpb (i2c-2)
Do you want to scan it? (yes/NO/selectively):

Next adapter: i915 gmbus dpd (i2c-3)
Do you want to scan it? (yes/NO/selectively):

Now follows a summary of the probes I have just done.
Just press ENTER to continue:

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

Do you want to overwrite /etc/sysconfig/lm_sensors? (YES/no):
Copy prog/init/lm_sensors.init to /etc/init.d/lm_sensors
for initialization at boot time.
You should now start the lm_sensors service to load the required
kernel modules.

Unloading i2c-dev... OK
```
**I will use windows system to check  whether the board supports wmi .And I can use ubuntu 19.10 with ‘sensors-detect’，because the kernel of unraid is 5.3**

**My English is poor, I hope you can understand, thank you!!!**

KeithMyers, 2019-10-22:
`Trying family VIA/Winbond/Nuvoton/Fintek...               Yes
Found unknown chip with ID 0xd121
    (logical device B has address 0x290, could be sensors)`

You probably have a Nuvoton NCT6793D Super I/O chip.  I would try and modprobe in the standard nct6775 driver.

[https://forums.linuxmint.com/viewtopic.php?t=221577](url)

Young-3, 2019-10-22:
So what should I do？

KeithMyers, 2019-10-22:
https://forums.linuxmint.com/viewtopic.php?t=221577[](url)
Well I would follow the steps at this link.
sudo modprobe -v nct6775
Then edit your /etc/modules/modules.conf file and add
nct6775 and save the file.  Then run sensors again and you should pick up everything your SIO chip exports.

Young-3, 2019-10-23:
**I follow the steps.But it seemed useless**

```
root@Tower:~# modprobe -v nct6775
root@Tower:~# sensors
acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

coretemp-isa-0000
Adapter: ISA adapter
CPU Temp:     +33.0°C  (high = +82.0°C, crit = +100.0°C)
Core 0:       +32.0°C  (high = +82.0°C, crit = +100.0°C)
Core 1:       +32.0°C  (high = +82.0°C, crit = +100.0°C)
Core 2:       +32.0°C  (high = +82.0°C, crit = +100.0°C)
Core 3:       +33.0°C  (high = +82.0°C, crit = +100.0°C)
Core 4:       +33.0°C  (high = +82.0°C, crit = +100.0°C)
Core 5:       +31.0°C  (high = +82.0°C, crit = +100.0°C)
```

```
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd121
    (logical device B has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
```

**Then I add ID: `modprobe nct 6775 force_id = 0xd121` . USELESS**

**lm-sensosrs version？Because I find 3.6 version support asus z370 some sensors**

Young-3, 2019-10-23:
Yes ,lm-sensors 3.6 version can detected 'Nuvoton NCT6793D Super IO Sensors'.So I need install 3.6 version.
```
root@Tower:~/lm-sensors-master/prog/detect# ./sensors-detect
# sensors-detect version 3.6.0+git
# Board: ASUSTeK COMPUTER INC. TUF Z370-PLUS GAMING
# Kernel: 5.3.6-Unraid x86_64
# Processor: Intel(R) Core(TM) i5-8400 CPU @ 2.80GHz (6/158/10)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no):
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no):
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found `Nuvoton NCT6793D Super IO Sensors'                   Success!
    (address 0x290, driver `nct6775')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no):
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (yes/NO):

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no):
Using driver `i2c-i801' for device 0000:00:1f.4: Kaby Lake (PCH)
Module i2c-dev loaded successfully.

Next adapter: SMBus I801 adapter at f040 (i2c-0)
Do you want to scan it? (YES/no/selectively):
Client found at address 0x51
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)
Client found at address 0x53
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)

Next adapter: i915 gmbus dpc (i2c-1)
Do you want to scan it? (yes/NO/selectively):

Next adapter: i915 gmbus dpb (i2c-2)
Do you want to scan it? (yes/NO/selectively):

Next adapter: i915 gmbus dpd (i2c-3)
Do you want to scan it? (yes/NO/selectively):

Now follows a summary of the probes I have just done.
Just press ENTER to continue:

Driver `nct6775':
  * ISA bus, address 0x290
    Chip `Nuvoton NCT6793D Super IO Sensors' (confidence: 9)

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

Do you want to generate /etc/sysconfig/lm_sensors? (yes/NO):
To load everything that is needed, add this to one of the system
initialization scripts (e.g. /etc/rc.d/rc.local):

#----cut here----
# Chip drivers
modprobe coretemp
modprobe nct6775
/usr/bin/sensors -s
#----cut here----

You really should try these commands right now to make sure everything
is working properly. Monitoring programs won't work until the needed
modules are loaded.

Unloading i2c-dev... OK
```

KeithMyers, 2019-10-23:
Just let sensors-detect add the commands for loading the nct6775 driver for you.  The reason your attempt failed is you didn't name the driver correctly.  You put a space between nct and 6775.  You shouldn't have to use a force ID parameter.  Not familiar with unraid or slack distro so don't know where you normally put driver modules to install at load time.  You still may have to use the enforce acpi_enforce_resources=lax kernel command.  I do on my ASUS X99-E-10G WS motherboard to not have the system ignore and not load the nct6775 driver.

Young-3, 2019-10-31:
Thanks，the problem has already been solved. I use the lasted sensors-detect and add this command `acpi_enforce_resources = lax` to boot. It's fine

electrified, 2019-10-31:
Glad you got it working :)

electrified, 2019-10-31:
And thanks @KeithMyers for answering in my absence :)

closed 2019-10-31 by electrified