electrified/asus-wmi-sensors #42: Question: Can this detect fan speeds on a header connect to the mobo by a 4 pin?

David-Frick, 2019-11-08:
I am trying to see if I can control fan speeds from the command line on my asus X399-A.
I have 4 or 5 case fans connects to a fan header that is then connected to the mobo by a 4 pin.
The output of the command `sensors` gives this:
```
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +27.4°C  (high = +70.0°C)
Tctl:         +54.4°C  

asuswmisensors-isa-0000
Adapter: ISA adapter
CPU Core Voltage:         +0.76 V  
CPU SOC Voltage:          +0.85 V  
DRAM AB Voltage:          +1.20 V  
DRAM CD Voltage:          +1.18 V  
1.8V PLL Voltage:         +1.79 V  
+12V Voltage:            +11.97 V  
+5V Voltage:              +4.96 V  
3VSB Voltage:             +3.31 V  
VBAT Voltage:             +3.05 V  
AVCC3 Voltage:            +3.31 V  
SB 1.05V Voltage:         +1.04 V  
CPU Core Voltage:         +0.76 V  
CPU SOC Voltage:          +0.83 V  
DRAM AB Voltage:          +1.20 V  
DRAM CD Voltage:          +1.20 V  
CPU Fan:                 2518 RPM
Chassis Fan 1:              0 RPM
Chassis Fan 2:              0 RPM
Chassis Fan 3:              0 RPM
Water Pump 1:               0 RPM
CPU OPT:                    0 RPM
EXT Fan 1:                  0 RPM
EXT Fan 2:                  0 RPM
EXT Fan 3:                  0 RPM
Cover Fan:                  0 RPM
CPU Temperature:          +27.0°C  
CPU Socket Temperature:   +27.0°C  
Motherboard Temperature:  +27.0°C  
Chipset Temperature:      +46.0°C  
Tsensor Temperature:     +216.0°C  
CPU VRM Temperature:      +29.0°C  
EXT Tsensor 1:             +0.0°C  
EXT Tsensor 2:             +0.0°C  
EXT Tsensor 3:             +0.0°C  
CPU VRM Output Current:   +1.00 A  

k10temp-pci-00cb
Adapter: PCI adapter
Tdie:         +27.0°C  (high = +70.0°C)
Tctl:         +54.0°C  
```
it does not seem to detect the fans on the fan header at all, or it seems to add them all up for the CPU fan (2500 RPM sounds awfully high). Do you have any advice on this by chance?

KeithMyers, 2019-11-08:
If a fan isn't discretely connected to a fan header, no fan rpm can be detected.  No tacho signal means no fan speed.  You have one fan connected to the cpu fan header that is being read.  Power connections are meaningless.  Unless you have the tacho wire from a 3 pin or 4 pin fan connected to a header, you can't detect any speed. If you want to control your fans, each fan has to be connected to a discrete fan header.  If you gang multiple fans together on a multi connector harness, only one fan in the harness is read and control for that one fan is applied to all the connected fans on the multi connector harness.

David-Frick, 2019-11-08:
I do run commands to set the fan speed for the detected cpu fan (which is
actually two) exhaust fans. I have 6 fans total, so I am still trying to
figure out how I want to run it. I have two front intake fans which I would
hope to keep the same speed and run at higher rpms than the others. I then
have 4 exhaust, 2 for the cpu (controllable), 1 for the gpu, and 1 at the
back of the case which will have a gpu in SLI sometime in the near future.
The GPU is an EVGA one and is connected to its fan header.

To be honest, I am not entirely sure how to proceed. I could easily move
the fans from the fan header to the motherboard itself as your post says,
which makes perfect sense. Where would you suggest going from there?

Thanks for the quick reply as well. Hope all is well.

Thanks,
David

KeithMyers, 2019-11-08:
Well I just plug in fans wherever convenient on the motherboard that the fan connections reach.  I then set up the fans in the BIOS for whatever speed or temp I want them to adjust to.  Then I don't have to worry about fan control.  I normally just disable all my fan controls in the BIOS and run all my fans at full DC speed since all my computers are doing heavy math crunching and need maximum cooling for the cpus and gpus.

I did have to move all my fans off the motherboard headers on the system with the 3900X because of bad BIOS bug that disabled all the fan headers randomly.  I moved them to a 10 port fan controller and even now have them powered that way even though the fan speed control bug in the BIOS has been fixed.  But the fan controller only allows reporting of one fan speed and controls all the fans off the one control fan.  But again I just run all the fans at 100% so no need for fan control just power.

David-Frick, 2019-11-08:
Okay, thanks for the clarification.

closed 2020-02-14 by electrified