electrified/asus-wmi-sensors #43: ASUS TUF X470-PLUS GAMING: Is it supported?

mgoppelt, 2019-12-06:
I built the module, but it won't load.

asuswmisensors: Vendor: ASUSTeK COMPUTER INC. Board: TUF X470-PLUS GAMING BIOS version: 5007 WMI version: 0
asuswmisensors: Unsupported board

it87 doesn't work either.

it87: Unsupported chip (DEVID=0x8665)

The board has the IT8665E chip.

Handle 0x0026, DMI type 34, 11 bytes
Management Device
	Description: ITE IT8665E
	Type: Other
	Address: 0x00000295
	Address Type: I/O Port

SIV64X uses it. SIV64X doesn't use WMI.

Does this mean that the board is not WMI capable?

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: TUF X470-PLUS GAMING
	Version: Rev X.0x
	Serial Number: 1812xx
	Asset Tag: Default string
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Default string
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x0022, DMI type 10, 6 bytes
On Board Device Information
	Type: Video
	Status: Enabled
	Description:    To Be Filled By O.E.M.

Handle 0x0027, DMI type 41, 11 bytes
Onboard Device
	Reference Designation:  Onboard IGD
	Type: Video
	Status: Enabled
	Type Instance: 1
	Bus Address: 0000:00:02.0

electrified, 2019-12-06:
The BIOS doesn't have the sensors WMI interface.

I downloaded the 5406 BIOS release and checked it in UEFItool like this https://github.com/electrified/asus-wmi-sensors/issues/11#issuecomment-515903663

and the ACPI methods needed aren't present.

electrified, 2019-12-17:
Closing this issue. If Asus add WMI support in a future BIOS please reopen.

closed 2019-12-17 by electrified