electrified/asus-wmi-sensors #44: Failed to build on kernel 5.4.2?

denglili, 2019-12-09:
I am on Ubuntu 18.04 and tried to upgraded to kernel 5.4.2 via ukuu, and asus-wmi-senors failed to build.

Currently rolled back to 5.3.15 and it works well.

KeithMyers, 2019-12-09:
Thanks for the warning.

electrified, 2019-12-09:
Can you paste the error message you are getting? The module builds OK on the current Arch Linux `5.4.2-arch1-1` kernel, so I don't think it is a general 5.4.2 issue.

denglili, 2019-12-10:
Here is the build log:
[make.log](https://github.com/electrified/asus-wmi-sensors/files/3943127/make.log)

electrified, 2019-12-10:
I think the version of GCC in 18.04 is too old to build the kernel (or any modules for it, hence the failure) due to changes in the header files in recent kernel versions.

Upgrading GCC to the version used to build the kernel would fix it, but I am not sure if this is possible (I'm not a debian/ubuntu expert)

See here for someone else having the same issue: https://bugs.launchpad.net/ubuntu/+source/fwts/+bug/1853090. Their fix is simply to not build the module.

denglili, 2019-12-10:
@electrified you are right, my GCC is too old. 18.04 has GCC 7 by default, switching to GCC 8 fixes the build. Thanks!

closed 2019-12-10 by denglili