electrified/asus-wmi-sensors #46: Fans detected by asuswmisensors-isa-0000 are uncontrolable

auyer, 2020-01-07:
I looked for solutions everywhere, but couldn't find it.
Here is the output for sensors -u and pwmconfig

```
[auyer@darkforce ~]$ sensors -u
k10temp-pci-00c3
Adapter: PCI adapter
Tdie:
  temp1_input: 48.625
  temp1_max: 70.000
Tctl:
  temp2_input: 48.625

asuswmisensors-isa-0000
Adapter: ISA adapter
CPU Core Voltage:
  in0_input: 1.406
+12V Voltage:
  in1_input: 12.164
+5V Voltage:
  in2_input: 4.987
3VSB Voltage:
  in3_input: 3.314
CPU Fan:
  fan1_input: 965.000
Chassis Fan 1:
  fan2_input: 941.000
Chassis Fan 2:
  fan3_input: 2070.000
Chassis Fan 3:
  fan4_input: 958.000
AIO Pump:
  fan5_input: 2419.000
Water Pump:
  fan6_input: 0.000
CPU OPT:
  fan7_input: 0.000
CPU Temperature:
  temp1_input: 48.000
Motherboard Temperature:
  temp2_input: 34.000
Chipset Temperature:
  temp3_input: 0.000
Tsensor 1 Temperature:
  temp4_input: 0.000

amdgpu-pci-0b00
Adapter: PCI adapter
vddgfx:
  in0_input: 0.800
fan1:
  fan1_input: 720.000
  fan1_min: 400.000
  fan1_max: 4900.000
edge:
  temp1_input: 41.000
  temp1_crit: 85.000
  temp1_crit_hyst: -273.150
  temp1_emergency: 90.000
junction:
  temp2_input: 43.000
  temp2_crit: 105.000
  temp2_crit_hyst: -273.150
  temp2_emergency: 110.000
mem:
  temp3_input: 42.000
  temp3_crit: 95.000
  temp3_crit_hyst: -273.150
  temp3_emergency: 100.000
power1:
  power1_average: 5.000
  power1_cap: 200.000

[auyer@darkforce ~]$ sudo pwmconfig 
# pwmconfig version 3.6.0
This program will search your sensors for pulse width modulation (pwm)
controls, and test each one to see if it controls a fan on
your motherboard. Note that many motherboards do not have pwm
circuitry installed, even if your sensor chip supports pwm.

We will attempt to briefly stop each fan using the pwm controls.
The program will attempt to restore each fan to full speed
after testing. However, it is ** very important ** that you
physically verify that the fans have been to full speed
after the program has completed.

Found the following devices:
   hwmon0 is asuswmisensors
   hwmon1 is asus
   hwmon2 is k10temp
   hwmon3 is amdgpu
   hwmon4 is hidpp_battery_0

Found the following PWM controls:
   hwmon3/pwm1           current value: 35
hwmon3/pwm1 is currently setup for automatic speed control.
In general, automatic mode is preferred over manual mode, as
it is more efficient and it reacts faster. Are you sure that
you want to setup this output for manual control? (n) 

```

electrified, 2020-01-07:
Hi there,

At the moment the driver does not have the ability to set fan speeds, the only thing I can suggest is setting fan curves in the UEFI.

The WMI interface contains some clearly named methods for reading sensor statuses, and then a load of methods with generic names. There is nothing obvious to set fan speeds. (see below)

I think there is a possibility that these generic methods are used by the Windows AISuite software, however we would need to discover which method to use. My attempts at determining what AISuite does by attaching a debugger and watching what API calls it makes have so far been unsuccessful.

The other option would be attempting to fix up the out of tree it87 driver to work correctly on current kernels. I had hoped that someone might have managed to do this, but I don't believe anyone has done it yet.

```
PS C:\WINDOWS\system32> get-WmiObject -class ASUSHW -Namespace "root\WMI" |  Get-Member -MemberType method

   TypeName: System.Management.ManagementObject#root\WMI\ASUSHW

Name                      MemberType Definition                                                                                                                                                          
----                      ---------- ----------                                                                                                                                                          
asio_hw_fun07             Method     System.Management.ManagementBaseObject asio_hw_fun07(System.UInt16 wPort)                                                                                           
asio_hw_fun08             Method     System.Management.ManagementBaseObject asio_hw_fun08(System.UInt16 wPort, System.Byte bData)                                                                        
asio_hw_fun09             Method     System.Management.ManagementBaseObject asio_hw_fun09(System.UInt32 dwAddr, System.Byte Width)                                                                       
asio_hw_fun10             Method     System.Management.ManagementBaseObject asio_hw_fun10(System.UInt32 dwAddr, System.Byte Width, System.UInt32 dwData)                                                 
asio_hw_fun11             Method     System.Management.ManagementBaseObject asio_hw_fun11(System.Byte Bus, System.Byte Dev, System.Byte Func, System.UInt16 Reg, System.Byte Width)                      
asio_hw_fun12             Method     System.Management.ManagementBaseObject asio_hw_fun12(System.Byte Bus, System.Byte Dev, System.Byte Func, System.UInt16 Reg, System.Byte Width, System.UInt32 dwData)
asio_hw_fun13             Method     System.Management.ManagementBaseObject asio_hw_fun13(System.UInt16 wIndexPort, System.Byte Index)                                                                   
asio_hw_fun14             Method     System.Management.ManagementBaseObject asio_hw_fun14(System.UInt16 wIndexPort, System.Byte Index, System.Byte Data)                                                 
asio_hw_fun15             Method     System.Management.ManagementBaseObject asio_hw_fun15(System.UInt32 Index)                                                                                           
asio_hw_fun16             Method     System.Management.ManagementBaseObject asio_hw_fun16(System.UInt32 Index, System.UInt32 DataHi, System.UInt32 DataLo)                                               
asio_hw_fun17             Method     System.Management.ManagementBaseObject asio_hw_fun17(System.UInt32 Index)                                                                                           
asio_hw_fun19             Method     System.Management.ManagementBaseObject asio_hw_fun19(System.Byte LDN, System.Byte Index)                                                                            
asio_hw_fun20             Method     System.Management.ManagementBaseObject asio_hw_fun20(System.Byte LDN, System.Byte Index, System.Byte Data)                                                          
asio_hw_fun21             Method     System.Management.ManagementBaseObject asio_hw_fun21(System.Byte Bank, System.Byte Index)                                                                           
asio_hw_fun22             Method     System.Management.ManagementBaseObject asio_hw_fun22(System.Byte Bank, System.Byte Index, System.Byte Data)                                                         
asio_hw_fun23             Method     System.Management.ManagementBaseObject asio_hw_fun23(System.String BankIndexArray)                                                                                  
asio_hw_fun24             Method     System.Management.ManagementBaseObject asio_hw_fun24(System.String BankIndexDataArray)                                                                              
asio_hw_fun25             Method     System.Management.ManagementBaseObject asio_hw_fun25(System.String BankIndexArray)                                                                                  
asio_hw_fun26             Method     System.Management.ManagementBaseObject asio_hw_fun26(System.String BankIndexDataArray)                                                                              
asio_hw_fun28             Method     System.Management.ManagementBaseObject asio_hw_fun28(System.Byte Index)                                                                                             
[... cut ...]
write_smbus_block         Method     System.Management.ManagementBaseObject write_smbus_block(System.UInt32 slave, System.UInt32 cmd, System.UInt32 count, System.Byte[] data)                           
write_smbus_byte          Method     System.Management.ManagementBaseObject write_smbus_byte(System.UInt32 slave, System.UInt32 cmd, System.UInt32 data)                                                 
write_smbus_word          Method     System.Management.ManagementBaseObject write_smbus_word(System.UInt32 slave, System.UInt32 cmd, System.UInt32 data)
```

auyer, 2020-01-08:
If I can help you running any tests, feel free to contact me.

snekku-dev, 2024-04-08:
Hello, any update on this? I am on an Asus ROG Strix B450-F Gaming and I believe I am encountering this same issue. When I run sensors-detect I get a warning telling me that "ITE IT8665E Super IO Sensors" doesn't have a driver in the kernel, which I believe might be the device capable of addressing the pwm values in the motherboard. This led me to the it87 module which I tried to load after removing the asus_wmi_sensors one, but without success since it outputs an error saying that there is no device for that module