electrified/asus-wmi-sensors #47: ASUS Rog Strix B450E Gaming - some readings are wrong.

asavah, 2020-01-09:
asus-wmi-sensors https://github.com/electrified/asus-wmi-sensors/commit/8daafd45d1b860cf5b17eee1c94d93feb04164a9
kernel 5.4.10
bios version 3004 2019/12/27
sensors version 3.6.0+git with libsensors version 3.6.0+git
```
asuswmisensors-isa-0000
Adapter: ISA adapter
CPU Core Voltage:
  in0_input: 0.807
VPP MEM Voltage:
  in1_input: 2.518
+12V Voltage:
  in2_input: 10.137
+5V Voltage:
  in3_input: 5.041
3VSB Voltage:
  in4_input: 3.292
VBAT Voltage:
  in5_input: 3.226
AVCC3 Voltage:
  in6_input: 3.292
SB 1.05V Voltage:
  in7_input: 1.031
CPU Core Voltage:
  in8_input: 0.000
CPU SOC Voltage:
  in9_input: 0.000
CPU Fan:
  fan1_input: 844.000
Chassis Fan 1:
  fan2_input: 0.000
Chassis Fan 2:
  fan3_input: 0.000
Chassis Fan 3:
  fan4_input: 0.000
AIO Pump:
  fan5_input: 0.000
Water Pump:
  fan6_input: 0.000
CPU OPT:
  fan7_input: 784.000
CPU Temperature:
  temp1_input: 30.000
CPU Socket Temperature:
  temp2_input: 36.000
Motherboard Temperature:
  temp3_input: 31.000
Chipset Temperature:
  temp4_input: 41.000
Tsensor 1 Temperature:
  temp5_input: 216.000
CPU VRM Temperature:
  temp6_input: 0.000
CPU VRM Output Current:
  curr1_input: 0.000
```

This one is wrong, in UEFI monitor it's 12.x-ish
+12V Voltage:
in2_input: 10.137

And this one can't be right, I'd probably smell it if it was :) .
Tsensor 1 Temperature:
  temp5_input: 216.000

KeithMyers, 2020-01-10:
> And this one can't be right, I'd probably smell it if it was :) .
> Tsensor 1 Temperature:
> temp5_input: 216.000
Perfectly normal and expected from an unterminated input on the motherboard.

electrified, 2020-01-10:
Hi.

As Keith says the 216 degrees means no sensor connected.

The 12V misreading is a BIOS issue on several of the B450 boards which Asus need to correct - can be seen in Windows too with apps that use WMI: https://www.hwinfo.com/forum/threads/low-readings-on-the-12v-line.6007/

asavah, 2020-01-10:
Thanks for the clarification and your awesome work.

closed 2020-01-10 by asavah