electrified/asus-wmi-sensors #49: CPU temperature reading stuck

pshemk, 2020-01-28:
I'm not sure if this is possible, but my CPU temperature reading (and only that single reading) is somehow stuck at 70°C:
[image]

I've tried rmmod and modprobe again but the reading doesn't change. I also checked the zenpower module and that one keeps reporting the temperature changes as expected.
This is on Asus X-470 Prime motherboard with AMD 3900x CPU.

electrified, 2020-02-08:
Are you using a version of the driver after this commit (Sept 2019)? https://github.com/electrified/asus-wmi-sensors/commit/8f5eaec0b29800f78999d8817d0c8fde14e978cd

PS is that Grafana? Nice work!

pshemk, 2020-02-09:
I'm using the current git version. After a reload the machine started to report correctly again.

And yes, that graph is from grafana :-)

electrified, 2020-02-09:
I think it's most likely that the problem is in the Asus firmware unfortunately :(.

Similar issues have been seen by other users, and also in Windows (see my comment here: https://github.com/electrified/asus-wmi-sensors/issues/48#issuecomment-578498632)

The WMI issues also seem more frequent on 3xxx gen CPUs.

You've inspired me to set up prometheus/node_exporter/grafana, I will monitor my own system more closely to see if I experience this issue.

closed 2020-02-09 by pshemk