electrified/asus-wmi-sensors #50: Add support for Asus Rog Zenith II Extreme

ylemoigne, 2020-02-02:
Hello,

Thanks for your work. I think that the Azus Rog Zenith II Extreme should be supported (but not sure of course).

Here the output of dmidecode
```
# dmidecode 3.1
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.
# SMBIOS implementations newer than version 3.1.1 are not
# fully supported by this version of dmidecode.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: ROG ZENITH II EXTREME
	Version: Rev 1.xx
	Serial Number: 191161971500294
	Asset Tag: Default string
	Features:
		Board is a hosting board
		Board is removable
		Board is replaceable
	Location In Chassis: Default string
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x003C, DMI type 10, 6 bytes
On Board Device Information
	Type: Video
	Status: Enabled
	Description:    To Be Filled By O.E.M.

Handle 0x0042, DMI type 41, 11 bytes
Onboard Device
	Reference Designation:  Onboard IGD
	Type: Video
	Status: Enabled
	Type Instance: 1
	Bus Address: 0000:00:02.0

Handle 0x0043, DMI type 41, 11 bytes
Onboard Device
	Reference Designation:  Onboard LAN
	Type: Ethernet
	Status: Enabled
	Type Instance: 1
	Bus Address: 0000:00:19.0

Handle 0x0044, DMI type 41, 11 bytes
Onboard Device
	Reference Designation:  Onboard 1394
	Type: Other
	Status: Enabled
	Type Instance: 1
	Bus Address: 0000:03:1c.2
```

electrified, 2020-02-02:
Hi there,

As with the X570 boards, ASUS has changed to using Nuvoton hardware monitoring ICs (specifically the NCT6798D) and dropped the WMI implementation on their recent Threadripper boards.

You need to use the `nct6775` kernel module. You may need an extra kernel param though: https://github.com/lm-sensors/lm-sensors/issues/197

ylemoigne, 2020-02-02:
Hi,

Thanks for the answer and link.

closed 2020-02-02 by ylemoigne