electrified/asus-wmi-sensors #52: TUF B450-PLUS GAMING support

gsedej, 2020-02-13:
Hello,

I tried this driver on my board and it looks like it's not supported
```
[feb13 16:27] asuswmisensors: Vendor: ASUSTeK COMPUTER INC. Board: TUF B450-PLUS GAMING BIOS version: 2008 WMI version: 0
[  +0,000000] asuswmisensors: Unsupported board
```
the Super IO is
```
iTE
IT8665E
1933-HXS
S17CLA
```
currently I can't test under windows.

the dmidecode
```
$ sudo dmidecode -t baseboard
# dmidecode 3.1
Getting SMBIOS data from sysfs.
SMBIOS 3.1.1 present.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: TUF B450-PLUS GAMING
	Version: Rev X.0x
	Serial Number: 191060797401531
	Asset Tag: Default string
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Default string
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x0020, DMI type 10, 6 bytes
On Board Device Information
	Type: Video
	Status: Enabled
	Description:    To Be Filled By O.E.M.

Handle 0x0025, DMI type 41, 11 bytes
Onboard Device
	Reference Designation:  Onboard IGD
	Type: Video
	Status: Enabled
	Type Instance: 1
	Bus Address: 0000:00:02.0

```

the chip does work using https://github.com/a1wong/it87 in kernel 5.6
```
git clone https://github.com/a1wong/it87
cd it87
make clean
sudo make install
sudo make dkms
sudo modprobe it87
```
read-out
```
$ sensors
it8665-isa-0290
Adapter: ISA adapter
in0:          +0.95 V  (min =  +1.72 V, max =  +2.33 V)  ALARM
in1:          +2.52 V  (min =  +0.34 V, max =  +2.08 V)  ALARM
in2:          +2.05 V  (min =  +2.77 V, max =  +2.67 V)  ALARM
in3:          +2.01 V  (min =  +0.32 V, max =  +2.60 V)
in4:          +0.03 V  (min =  +1.38 V, max =  +2.07 V)  ALARM
in5:          +0.03 V  (min =  +0.00 V, max =  +0.89 V)
in6:          +0.03 V  (min =  +2.51 V, max =  +2.67 V)  ALARM
3VSB:         +3.36 V  (min =  +5.08 V, max =  +1.57 V)  ALARM
Vbat:         +3.27 V  
+3.3V:        +3.36 V  
fan1:        1201 RPM  (min =   24 RPM)
fan2:           0 RPM  (min =   17 RPM)
fan3:           0 RPM  (min =   31 RPM)
fan4:           0 RPM  (min =   -1 RPM)
fan6:           0 RPM  (min =   -1 RPM)
temp1:        +38.0°C  (low  =  -3.0°C, high =  -9.0°C)  ALARM
temp2:        +33.0°C  (low  = -61.0°C, high = -18.0°C)  ALARM  sensor = thermistor
temp3:        +35.0°C  (low  = -83.0°C, high =  -8.0°C)  ALARM  sensor = thermistor
temp4:        +35.0°C  (low  =  -4.0°C, high = +121.0°C)  sensor = thermistor
temp5:        +35.0°C  (low  = -26.0°C, high = -41.0°C)  ALARM  sensor = thermistor
temp6:        +35.0°C  (low  = -13.0°C, high = -61.0°C)  ALARM  sensor = thermistor
intrusion0:  ALARM
```

electrified, 2020-02-14:
I checked the latest BIOS for this board (TUF-B450-PLUS-GAMING-ASUS-2008.CAP) and it doesn't have the WMI sensors interface.

Reopen this issue if it is added in a future BIOS release.

See this comment for how to check BIOS images
https://github.com/electrified/asus-wmi-sensors/issues/11#issuecomment-515903663

closed 2020-02-14 by electrified

OhSoGood, 2020-11-09:
@electrified  : if I do a "dmesg |grep asus", I get this:
```
asus_wmi: ASUS WMI generic driver loaded
asus_wmi: Initialization: 0x0
asus_wmi: BIOS WMI version: 0.9
asus_wmi: SFUN value: 0x0

```
Yes, compiling and installing your code, I also get 
`asuswmisensors: Vendor: ASUSTeK COMPUTER INC. Board: TUF B450M-PLUS GAMING BIOS version: 1804 WMI version: 0`

Isn't it contradictory?

electrified, 2020-11-09:
Hi there. The WMI sensors stuff has it's own version number which is an integer. I believe version numbers 1-4 are valid versions of the API that have been seen.

WMI sensors wasn't present in the 2008 UEFI image I tested in the previous message, it could be worth testing the latest 2202 as per https://github.com/electrified/asus-wmi-sensors/issues/11#issuecomment-515903663, although it's very unlikely they've added WMI sensors support.

OhSoGood, 2020-11-22:
Indeed, WMI sensors are not found (looking for the SENX keyword).