electrified/asus-wmi-sensors #55: Prime X470 Pro fans get stuck

sloth1, 2020-02-20:
I am seeing my fans get stuck at a fixed speed about 15-30 minutes after booting. Apparently this is a known problem if you read the sensors to often. Which I have been reading them non-stop to see how well my cooler is working. Buggy monitoring chip:
https://www.reddit.com/r/Amd/comments/8iykdf/the_asus_prime_x470_pro_has_been_nothing_more/

Do you know anything about this or any workaround? Scary when my fan gets stuck at 1300rpm and the CPU starts cooking.

KeithMyers, 2020-02-20:
Haven't heard of any issues from users that I have recommended the motherboard.

Do you have the latest BIOS?

sloth1, 2020-02-21:
Yes 5406. I will experiment some more.

sloth1, 2020-02-22:
I see that I have an asus_wmi module already (running openSUSE Tumblweed). Would that cause problems?

KeithMyers, 2020-02-22:
No I have the same asus_wmi modele installed also that supports other stuff.

Doesn't interfere with asus_wmi_sensors.

I don't control the fans by the motherboard for the most part now since ASUS for about half a year had broken BIOS' that would just shut off motherboard fan headers completely.  That instantly lead to overheated computers.  I went with a SATA powered fan hub after that fiasco.

ASUS has since fixed the BIOS', at least on my C7H boards that supposedly doesn't flub up the fans but I am still using the fan controller since it is independent of the motherboard completely.  Not worth switching every fan back to the mobo headers.

sloth1, 2020-02-26:
ok i think my problems may have been caused by a bad RAM stick. I will watch this a bit longer and if I can't reproduce I will close the issue. Thanks for making this sensor module!

electrified, 2020-02-27:
This boards WMI implementation seems to be particularly buggy, there have been at least 3 previous reports of issues: https://github.com/electrified/asus-wmi-sensors/issues/49 https://github.com/electrified/asus-wmi-sensors/issues/48 https://github.com/electrified/asus-wmi-sensors/issues/54

I would say it is probably coincidental that the problem has gone away....

sloth1, 2020-02-28:
yup. It did it again. I ran for 24 hours with the module loaded and reading every 0.5 seconds. My fan curve actually inverted or something.. fan speed went very high, I put a load on the CPU, fan speed came down as temps went up.

Then I rebooted and ran for 24 hours with no module. Everything was normal.

KeithMyers, 2020-02-28:
Maybe you could try the old out of band it87 driver as a substitute.  Still probably best action is to get ASUS correct the problems with that boards BIOS.

closed 2020-02-29 by sloth1