electrified/asus-wmi-sensors #56: ASUS ROG STRIX X470-I GAMING appears to be supported

jsho98, 2020-02-25:
I just installed this and it seems to be working perfectly on this board. I'm not sure what the minimum BIOS version is but I am running the latest version, 3004.
```
asuswmisensors-isa-0000
Adapter: ISA adapter
CPU Core Voltage:         +1.36 V  
VPP MEM Voltage:          +2.50 V  
+12V Voltage:            +10.08 V  
+5V Voltage:              +5.01 V  
3VSB Voltage:             +3.31 V  
VBAT Voltage:             +3.21 V  
AVCC3 Voltage:            +3.33 V  
SB 1.05V Voltage:         +1.05 V  
CPU Core Voltage:         +1.38 V  
CPU SOC Voltage:          +1.08 V  
CPU Fan:                 2064 RPM
Chassis Fan 1:              0 RPM
AIO Pump:                   0 RPM
CPU OPT:                    0 RPM
CPU Temperature:          +35.0°C  
CPU Socket Temperature:   +42.0°C  
Motherboard Temperature:  +37.0°C  
Chipset Temperature:      +62.0°C  
Tsensor 1 Temperature:   +216.0°C  
CPU VRM Temperature:      +45.0°C  
CPU VRM Output Current:   +6.00 A  
```
Just wanted to let you know so it can be added as a supported modle.

KeithMyers, 2020-02-25:
Thanks for the review.  Ed will appreciate adding it to the "known good" list.

electrified, 2020-02-27:
The motherboard identification string was in the driver, but the README hadn't been updated. I've fixed this, thanks

closed 2020-02-27 by electrified