electrified/asus-wmi-sensors #57: fails to build on custom kernel

TeamFahQ, 2020-02-27:
Your module works great on stock kernel, but I prefer to use my own kernel. I'm on kernel 5.3.13-lz-xan9-lz. Which is based off xanmod's linux kernel with some of my own tweaks. I initially compiled and install the module in stock 5.3.0-40-generic. Everthing compiled and installed without a problem. I then compiled my own build, and tried to install it but install failed. I then rebooted in to stock kernel, removed the module and built, rebooted, compile and installed kernel, rebooted, tried to run wmi module and it fails with the following
Error! Bad return status for module build on kernel: 5.3.13-lz-xan9-lz (x86_64)
And looking at the make.log
DKMS make.log for asus-wmi-sensors-8daafd4 for kernel 5.3.13-lz-xan9-lz (x86_64)
Thu Feb 27 14:05:25 CST 2020
make[3]: *** No rule to make target '/var/lib/dkms/asus-wmi-sensors/8daafd4/build/asus-wmi-sensors.o', needed by '__build'.  Stop.
Makefile:1628: recipe for target '_module_/var/lib/dkms/asus-wmi-sensors/8daafd4/build' failed
make[2]: *** [_module_/var/lib/dkms/asus-wmi-sensors/8daafd4/build] Error 2
Makefile:72: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
I looked at my .config and have
CONFIG_ACPI_WMI=y
CONFIG_ASUS_WMI=m
Any ideas as to what I may be missing?

electrified, 2020-02-27:
How are you building and installing the kernel? 

What does `dkms status` say?

It's been a while since I used DKMS, but it creates symlinks into the kernel source tree from under `/var/lib/dkms/`. I'd check that there aren't any broken symlinks under `/var/lib/dkms/asus-wmi-sensors/` and uninstall the module via dkms and reinstall.

TeamFahQ, 2020-02-29:
For building/installing I use a debian method from [here](https://scottlinux.com/2011/07/15/how-to-compile-your-own-kernel-in-ubuntu-debian-mint/)
My source is in my home partition, but it is properly linked in /usr/src (when it's installed)
`$ dkms status
asus-wmi-sensors, 8daafd4: added
bcmwl, 6.30.223.271+bdcom, 5.3.0-28-generic, x86_64: installed
bcmwl, 6.30.223.271+bdcom, 5.3.0-40-generic, x86_64: installed`
As you can see, I don't have my kernel installed at this time.
I do have a broken link in `/var/lib/dkms/asus-wmi-sensors/
Name: kernel-5.3.0-40-generic-x86_64
Link target: 8daafd4/5.3.0-40-generic/x86_64`
I ran sudo make dkms -i and it fixed the broken link though.
I copied asus-wmi-sensors.ko to drivers/hwmon in my kernel source and ran a build. Installed the kernel but still got an error.
`Error! Bad return status for module build on kernel: 5.3.13-lz-xan9-lz (x86_64)
Consult /var/lib/dkms/asus-wmi-sensors/8daafd4/build/make.log for more information.`
How can I uninstall this driver and start over? Maybe I should add the whole thing to my source?

TeamFahQ, 2020-03-06:
I finally got it. I uninstalled my kernel and booted in with kernel 5.3-40-generic. I build the module, performed a clean build on my kernel with a fresh and new .config, and installed it. Booted with my kernel, and copied the entire asus-wmi-sensors folder to

`/lib/modules/5.3.13-lz/kernel/drivers/hwmon`. I also added

` kernel/drivers/hwmon/asus-wmi-sensors/asus-wmi-sensors.ko: kernel/drivers/platform/x86/wmi.ko` 

to the modules.dep file.

Rebooted and the ran

`sudo make dkms`

The module built as it should and I was able to modprobe it. Added it to my /etc/modules list so it would load at start up, rebooted and all is working flawlessly.

KeithMyers, 2020-03-06:
Good effort for tenacity and persistence. Congratz.

electrified, 2020-03-06:
If you've incorporated the module into your kernel source tree line an in-tree module (e.g. added it to `drivers/hwmon/Kconfig` and `drivers/hwmon/Makefile` and configured it on) , the module should be built with the kernel and you don't need DKMS, as it's only purpose is to build out-of-tree modules when new kernels are installed. 

If installing packaged kernels, DKMS is triggered after the new kernel package has been installed, from the `No rule to make target ` errors it sounds like perhaps DKMS is being run at the wrong time, or the manual build process is otherwise interfering with it.

Another options is to not use DKMS and just `sudo make` the module once you have booted into your new kernel which builds the module as per https://www.kernel.org/doc/html/latest/kbuild/modules.html

Closing as you have resolved the issue.

closed 2020-03-06 by electrified

TeamFahQ, 2020-03-22:
If anyone else is encountering this issue, here is my commit to incorporate the asus-wmi-sensors into your kernel source. [Build asus-wmi-sensors with kernel](https://github.com/TeamFahQ/linux-lz_kernel/commit/33649444a9934e26b0c4cf91e7bede62ec5afc5a)