electrified/asus-wmi-sensors #58: If build with no dkms, breaks build even after install dkms

dtgriscom, 2020-02-29:
I'm using Ubuntu 18.04.4. I cloned this repository, and tried

`sudo make dkms`

But, got an error:

````
make: dkms: Command not found
Makefile:91: recipe for target 'dkms' failed
make: *** [dkms] Error 127
````

I installed `dkms`, and then tried to make this again, but got a different error:

````
mkdir: cannot create directory ‘/usr/src/asus-wmi-sensors-b6c25d2’: File exists
Makefile:91: recipe for target 'dkms' failed
make: *** [dkms] Error 1
````

So, the first failed make had created the directory, but subsequent makes fail if the directory exists. I tried `make clean`, but it didn't remove the created directory. I finally hand-deleted the directory, and the `sudo make dkms` got a lot further (it ended up failing, but that'll be a different report).

Suggestions:

- Have `sudo make clean` remove the source directory
- Check for `dkms` (and other prerequisites?) before creating the source directory

KeithMyers, 2020-02-29:
I've fallen into this Catch-22 myself a few times.  Always had to remove the flubbed original try at the dkms install before being successful.

TeamFahQ, 2020-02-29:
Try `sudo make dkms -i`

The -i flag will ignore the file exists error

dtgriscom, 2020-02-29:
Telling make to ignore all errors wouldn't feel safe to me as an end-user; it would be far better to have the Makefile itself be more robust.

Given that the directory name is built using `git describe --long --always`, there's virtually no chance that automatically deleting the directory would cause harm, right?

TeamFahQ, 2020-02-29:
I see your point. But it was just a suggestion. I tried deleting the directories, but I'm not sure where it installs to, and in doing so, I caused the same error you are getting. I used the -i flag, and it built fine and fixed the one symlink I found that was broken. But it still didn't fix my issue...

electrified, 2020-03-06:
I will add a check to the makefile to ensure DKMS is installed.