electrified/asus-wmi-sensors #59: Random system shutdowns

pshemk, 2020-03-05:
I don't think it's an issue with asus-wmi-sensors, but with the actual BIOS, but I'm posting it here as a reference if someone encounters similar problems.

I initially had an issue with the CPU overheating and shutting down. I was not monitoring temperatures at that time. Then I found this project and started to monitor the temperature and could see that the CPU was overheating (system was shutting down when it reached over 105°C). I fixed cooling and things improved, but I've noticed that the system was also shutting down when almost completely idle, logs and temperature readings didn't indicate any problems. I have a PRIME X470-PRO motherboard and googling revealed that users of HWiNFO (which also uses the WMI interface) also had those problems. I initially reduced the frequency of pooling (from 10s to 2mins) to see if that makes a difference, but still the best uptime I've achieved was around 4-5days and it didn't feel like this made a difference. Ultimately over 3 weeks ago I've disabled the wmi module completely. Since then I didn't have a single random shutdown, even under very heavy load.

electrified, 2020-03-05:
Hi.

This boards WMI implementation seems to be particularly buggy, there have been now been 4 previous reports of issues: #49 #48 #54 #55.

As you say, this is down to the BIOS implementation.

Debating whether to remove support for it until Asus fix the BIOS...

KeithMyers, 2020-03-06:
The fan headers shutting off was a problem in the C7H BIOS for about 6 months.  I ended up taking all fans off the motherboard header and using a offboard fan controller to power the fans.  I could not depend on the motherboard to just stop outputting 12V and my cpu would overheat.

They finally fixed the BIOS in the 2901 BIOS and reports are that it is still working in the current 3004 BIOS.  But I continue to use the fan controller so don't have any worries if the BIOS breaks again.

electrified, 2020-05-13:
Closing as fix requires a BIOS change from ASUS

closed 2020-05-13 by electrified