electrified/asus-wmi-sensors #60: Failed to build it via AUR.

laichiaheng, 2020-03-15:
==> dkms install asus-wmi-sensors/53.b6c25d2 -k 5.5.9-1-MANJARO
Error! Bad return status for module build on kernel: 5.5.9-1-MANJARO (x86_64)

electrified, 2020-03-15:
Hey.

I have just tested this, and it works OK with the `linux55` kernel, which for me currently gets 5.5.8-1-MANJARO.

Installed from `manjaro-kde-19.0.2-200311-linux54.iso`, updated, installed kernel `linux55`, and `linux55-headers`.

Installed with `pamac build asus-wmi-sensors-dkms-git` and all worked OK.

What is in the DKMS logs?

laichiaheng, 2020-03-16:
How to check the DKMS logs?
It used to work with 5.5.9 for me, too.

laichiaheng, 2020-03-16:
It works again after Manjaro release a fix for 5.5.9

closed 2020-03-16 by laichiaheng

electrified, 2020-03-16:
Glad you have got it sorted :)