electrified/asus-wmi-sensors #61: Cover fan and AIO pump not reported in output ROG Strixx X399-E Gaming

TeamFahQ, 2020-03-22:
Temp sensors are working great. But Cover fan and AIO pump aren't being reported in psensor. There is an option for water pump, but this board has 2 water pump headers. Header1 is for a normal pwm style pump, whereas Header2 is for an AIO 3 pin pump. For the cover fan, it is reporting, but not showing an output. Here's a [Screenshot](https://drive.google.com/open?id=1-1BzzZ3jFmofhgSl6GCJ6DQF4Z4ZhjI_).

KeithMyers, 2020-03-22:
Wasn't sure what you were calling "cover" fan.  I first thought you were referring to the fan on all the X570 PCH covers, but realized you were referring to an old X399 platform.

Guessing "cover" fan refers to the small fan in the I/O shield for the VRM cooling.

That fan might be going through one of ASUS' typical proprietary fan controller chips and not exported as WMI data that the driver can use.

KeithMyers, 2020-03-22:
Off topic.  Can you tell me what application is running in Terminal?  And what is the name of the desktop monitor app?

TeamFahQ, 2020-03-22:
> Wasn't sure what you were calling "cover" fan. I first thought you were referring to the fan on all the X570 PCH covers, but realized you were referring to an old X399 platform.
>
> Guessing "cover" [...]

Sorry, should've been more specific. Yes, I am refering to the small fan in the I/O area for cooling VRM.

TeamFahQ, 2020-03-22:
> Off topic. Can you tell me what application is running in Terminal? And what is the name of the desktop monitor app?

Desktop monitor app is conky (with my own configuration). I use s-tui to stress test cpu.

KeithMyers, 2020-03-22:
Thanks.  I thought that might be Conky with a custom theme.  I'll have to investigate s-tui.

electrified, 2020-03-22:
Hi there.

This would need to be fixed in the BIOS.

What I would suggest is recreating the issue in Windows with HWiNFO, AIDA64 or similar software that uses the WMI interface, and who's author has contacts in ASUS. They may be able to pass the message along to ASUS.

Contacting ASUS support directly usually results in being ignored unfortunately (or they tell you to use AI Suite)

TeamFahQ, 2020-03-23:
> Hi there.
>
> This would need to be fixed in the BIOS.
>
> What I would suggest is recreating the issue in Windows with HWiNFO, AIDA64 or similar software that uses the WMI interface, and who's author has [...]

That's just it, the cover fan speed and AIO speed IS being reported in windows (AI Suite). I can physically see the cover fan spinning and if I put my hand on the pump, I can feel it moving. I can also monitor and adjust the curve on both in BIOS (UEFI) and AI Suite. Could it possibly be that the X399-E has a different address/location/class for these devices?
If so, how would one go about getting that information?

electrified, 2020-03-23:
Hey,

AI Suite doesn't use WMI sensors. That's why I said ASUS will say to use it. They know WMI is buggy.

Use HWiNFO, AIDA , SIV etc as they use WMI. You should see the same issue.

electrified, 2020-03-23:
Also note that most boards have some oddity in the WMI sensors output VS the BIOS. On my Crosshair 7, it's not possible to see the Extension Fan value headers despite them being in the BIOS and (AI Suite). Other boards display the 12V value with an offset of -2V. Others like yours have some values stuck at 0/ missing.

Sensors WMI is a dead end - Elmor has left ASUS (it was his idea), and the newer X570 and X499 boards use a Nuvoton sensors IC and don't have WMI in their BIOSes. ASUS' BIOS releases on older boards have slowed dramatically so it is unlikely fixes will be forthcoming.

The only real workaround would be for someone to fix up the out-of-tree IT87 driver, and bypass ASUS kludgy WMI. This is unlikely to happen, as the reason Gunther Roeck stopped maintaining it was it is not possible to get it to work reliably, and there are no publically available datasheets for the more recent sensors ICs.

TeamFahQ, 2020-03-24:
> Also note that most boards have some oddity in the WMI sensors output VS the BIOS. On my Crosshair 7, it's not possible to see the Extension Fan value headers despite them being in the BIOS and (AI [...]

A lot of info I didn't know. I found an old modified asus_wmi. I wonder if I could somehow bring it a little more up to date? It's for fan control... But not having a data sheet on the board makes it nearly impossible.

electrified, 2020-05-13:
Closing as fix requires a BIOS change from ASUS.

closed 2020-05-13 by electrified