electrified/asus-wmi-sensors #63: add support for board ROG Zenith II Extreme

krzemienski, 2020-04-08:
```
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: ROG ZENITH II EXTREME
	Version: Rev 1.xx
	Serial Number: 191161234900127
	Asset Tag: Default string
	Features:
		Board is a hosting board
		Board is removable
		Board is replaceable
	Location In Chassis: Default string
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x003C, DMI type 10, 6 bytes
On Board Device Information
	Type: Video
	Status: Enabled
	Description:    To Be Filled By O.E.M.

Handle 0x0042, DMI type 41, 11 bytes
Onboard Device
	Reference Designation:  Onboard IGD
	Type: Video
	Status: Enabled
	Type Instance: 1
	Bus Address: 0000:00:02.0

Handle 0x0043, DMI type 41, 11 bytes
Onboard Device
	Reference Designation:  Onboard LAN
	Type: Ethernet
	Status: Enabled
	Type Instance: 1
	Bus Address: 0000:00:19.0

Handle 0x0044, DMI type 41, 11 bytes
Onboard Device
	Reference Designation:  Onboard 1394
	Type: Other
	Status: Enabled
	Type Instance: 1
	Bus Address: 0000:03:1c.2

```

Currently when trying to install via make dkms I run into this error, Is this due to the board not being found in the support boards map it looks like is used? Hopefully it is as simple as that, I wasn't sure how/if the define you have the board name needed a specific naming convention. 

```
Creating symlink /var/lib/dkms/asus-wmi-sensors/b6c25d2/source ->
                 /usr/src/asus-wmi-sensors-b6c25d2

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j64 KERNELRELEASE=5.6.2-050602-generic TARGET=5.6.2-050602-generic...
Signing module:
 - /var/lib/dkms/asus-wmi-sensors/b6c25d2/5.6.2-050602-generic/x86_64/module/asus-wmi-sensors.ko
Secure Boot not enabled on this system.
cleaning build area...

DKMS: build completed.

asus-wmi-sensors.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.6.2-050602-generic/updates/dkms/

depmod...

DKMS: install completed.
modprobe: ERROR: could not insert 'asus_wmi_sensors': No such device
make: *** [Makefile:102: dkms] Error 1
```

electrified, 2020-04-08:
Hi.

The Asus Threadripper 3rd gen boards, like the X570 boards don't have the WMI interface.

They do however have the Nuvoton NCT6798D sensor IC which is supported by the mainline  `nct6775` driver.

krzemienski, 2020-04-08:
Well that was really freaking simple... 

just ran `sudo sensors-detect —auto`

Thanks for you work on this project though, maybe one day I’ll be as brave as you and write kernel modules for fun!

closed 2020-04-08 by krzemienski