electrified/asus-wmi-sensors #64: ERROR: could not insert 'asus_wmi_sensors': No such device

belkone, 2020-04-22:
Hi, 
Is it possible to run it for below one ?
```
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.1.1 present.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: PRIME B450M-A
	Version: Rev X.0x
	Serial Number: [masked]
	Asset Tag: Default string
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Default string
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x0022, DMI type 10, 6 bytes
On Board Device Information
	Type: Video
	Status: Enabled
	Description:    To Be Filled By O.E.M.

Handle 0x0027, DMI type 41, 11 bytes
Onboard Device
	Reference Designation:  Onboard IGD
	Type: Video
	Status: Enabled
	Type Instance: 1
	Bus Address: 0000:00:02.0
```

```
$ dmidecode -t 34

# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.1.1 present.

Handle 0x0026, DMI type 34, 11 bytes
Management Device
	Description: ITE IT8655E
	Type: Other
	Address: 0x00000295
	Address Type: I/O Port
```

compilation goes correctly, the problem is only with modprobe (I also tried to add to this source my mobo name/change version by myself to cheat board incompatibility;p) :

```
$ modprobe asus-wmi-sensors -vvv
modprobe: INFO: ../libkmod/libkmod.c:364 kmod_set_log_fn() custom logging function 0x56007ec55150 registered
modprobe: DEBUG: ../libkmod/libkmod-index.c:755 index_mm_open() file=/lib/modules/5.3.18-3-pve/modules.dep.bin
modprobe: DEBUG: ../libkmod/libkmod-index.c:755 index_mm_open() file=/lib/modules/5.3.18-3-pve/modules.alias.bin
modprobe: DEBUG: ../libkmod/libkmod-index.c:755 index_mm_open() file=/lib/modules/5.3.18-3-pve/modules.symbols.bin
modprobe: DEBUG: ../libkmod/libkmod-index.c:755 index_mm_open() file=/lib/modules/5.3.18-3-pve/modules.builtin.bin
modprobe: DEBUG: ../libkmod/libkmod-module.c:556 kmod_module_new_from_lookup() input alias=asus-wmi-sensors, normalized=asus_wmi_sensors
modprobe: DEBUG: ../libkmod/libkmod-module.c:562 kmod_module_new_from_lookup() lookup modules.dep asus_wmi_sensors
modprobe: DEBUG: ../libkmod/libkmod.c:574 kmod_search_moddep() use mmaped index 'modules.dep' modname=asus_wmi_sensors
modprobe: DEBUG: ../libkmod/libkmod.c:402 kmod_pool_get_module() get module name='asus_wmi_sensors' found=(nil)
modprobe: DEBUG: ../libkmod/libkmod.c:410 kmod_pool_add_module() add 0x560080b1b070 key='asus_wmi_sensors'
modprobe: DEBUG: ../libkmod/libkmod.c:402 kmod_pool_get_module() get module name='wmi' found=(nil)
modprobe: DEBUG: ../libkmod/libkmod.c:402 kmod_pool_get_module() get module name='wmi' found=(nil)
modprobe: DEBUG: ../libkmod/libkmod.c:410 kmod_pool_add_module() add 0x560080b1b250 key='wmi'
modprobe: DEBUG: ../libkmod/libkmod-module.c:196 kmod_module_parse_depline() add dep: /lib/modules/5.3.18-3-pve/kernel/drivers/platform/x86/wmi.ko
modprobe: DEBUG: ../libkmod/libkmod-module.c:202 kmod_module_parse_depline() 1 dependencies for asus_wmi_sensors
modprobe: DEBUG: ../libkmod/libkmod-module.c:583 kmod_module_new_from_lookup() lookup asus_wmi_sensors=0, list=0x560080b1ada0
modprobe: DEBUG: ../libkmod/libkmod.c:501 lookup_builtin_file() use mmaped index 'modules.builtin' modname=asus_wmi_sensors
modprobe: DEBUG: ../libkmod/libkmod-module.c:1750 kmod_module_get_initstate() could not open '/sys/module/asus_wmi_sensors/initstate': No such file or directory
modprobe: DEBUG: ../libkmod/libkmod-module.c:1760 kmod_module_get_initstate() could not open '/sys/module/asus_wmi_sensors': No such file or directory
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=snd_pcsp mod->name=wmi mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=snd_usb_audio mod->name=wmi mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=cx88_alsa mod->name=wmi mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=snd_atiixp_modem mod->name=wmi mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=snd_intel8x0m mod->name=wmi mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=snd_via82xx_modem mod->name=wmi mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=kvm mod->name=wmi mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=kvm_amd mod->name=wmi mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=bonding mod->name=wmi mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=dummy mod->name=wmi mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod.c:501 lookup_builtin_file() use mmaped index 'modules.builtin' modname=wmi
modprobe: DEBUG: ../libkmod/libkmod-module.c:1306 kmod_module_probe_insert_module() Ignoring module 'wmi': already loaded
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=snd_pcsp mod->name=asus_wmi_sensors mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=snd_usb_audio mod->name=asus_wmi_sensors mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=cx88_alsa mod->name=asus_wmi_sensors mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=snd_atiixp_modem mod->name=asus_wmi_sensors mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=snd_intel8x0m mod->name=asus_wmi_sensors mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=snd_via82xx_modem mod->name=asus_wmi_sensors mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=kvm mod->name=asus_wmi_sensors mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=kvm_amd mod->name=asus_wmi_sensors mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=bonding mod->name=asus_wmi_sensors mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1393 kmod_module_get_options() modname=dummy mod->name=asus_wmi_sensors mod->alias=(null)
modprobe: DEBUG: ../libkmod/libkmod-module.c:1750 kmod_module_get_initstate() could not open '/sys/module/asus_wmi_sensors/initstate': No such file or directory
modprobe: DEBUG: ../libkmod/libkmod-module.c:1760 kmod_module_get_initstate() could not open '/sys/module/asus_wmi_sensors': No such file or directory
modprobe: DEBUG: ../libkmod/libkmod-module.c:744 kmod_module_get_path() name='asus_wmi_sensors' path='/lib/modules/5.3.18-3-pve/updates/dkms/asus-wmi-sensors.ko'
modprobe: DEBUG: ../libkmod/libkmod-module.c:744 kmod_module_get_path() name='asus_wmi_sensors' path='/lib/modules/5.3.18-3-pve/updates/dkms/asus-wmi-sensors.ko'
insmod /lib/modules/5.3.18-3-pve/updates/dkms/asus-wmi-sensors.ko 
modprobe: DEBUG: ../libkmod/libkmod-module.c:744 kmod_module_get_path() name='asus_wmi_sensors' path='/lib/modules/5.3.18-3-pve/updates/dkms/asus-wmi-sensors.ko'
modprobe: INFO: ../libkmod/libkmod-module.c:886 kmod_module_insert_module() Failed to insert module '/lib/modules/5.3.18-3-pve/updates/dkms/asus-wmi-sensors.ko': No such device
modprobe: ERROR: could not insert 'asus_wmi_sensors': No such device
modprobe: DEBUG: ../libkmod/libkmod-module.c:468 kmod_module_unref() kmod_module 0x560080b1b070 released
modprobe: DEBUG: ../libkmod/libkmod.c:418 kmod_pool_del_module() del 0x560080b1b070 key='asus_wmi_sensors'
modprobe: DEBUG: ../libkmod/libkmod-module.c:468 kmod_module_unref() kmod_module 0x560080b1b250 released
modprobe: DEBUG: ../libkmod/libkmod.c:418 kmod_pool_del_module() del 0x560080b1b250 key='wmi'
modprobe: INFO: ../libkmod/libkmod.c:331 kmod_unref() context 0x560080b1a480 released
```

kernel: 5.3.18-3

is there anything I can try ?

electrified, 2020-04-30:
Hi there,

I've checked the BIOS image (PRIME-B450M-A-ASUS-2006.CAP) using the method in https://github.com/electrified/asus-wmi-sensors/issues/11#issuecomment-515903663 and unfortunately this board doesn't have the WMI sensors interface in it's BIOS.

belkone, 2020-04-30:
Ok thank you for help!
I hadn't know before how to check the existence of this sensor for this mobo, i'm closing issue, thanks!

closed 2020-04-30 by belkone