electrified/asus-wmi-sensors #65: build failed due to missing linux/wmi.h

cade-vs, 2020-05-09:
hello!

just tried to build current git snapshot (commit b6c25d2edeb) with kernel 4.9.0-12-amd64 (Devuan) and got this:

-----------------
DKMS make.log for asus-wmi-sensors-b6c25d2 for kernel 4.9.0-12-amd64 (x86_64)
Sun May 10 00:57:55 EEST 2020
  CC [M]  /var/lib/dkms/asus-wmi-sensors/b6c25d2/build/asus-wmi-sensors.o
/var/lib/dkms/asus-wmi-sensors/b6c25d2/build/asus-wmi-sensors.c:18:23: fatal error: linux/wmi.h: No such file or directory
 #include <linux/wmi.h>
                       ^
compilation terminated.
-----------------

I did "git clone" and "make dkms -i", am I missing something?

thanks.

electrified, 2020-05-11:
Hi there.

It looks like your kernel doesn't have the WMI headers. The WMI support we use was introduced in kernel 4.12 (July 2017). Some of the RedHat kernels have backported the WMI support to kernels older than that, but I don't think Debian have.

cade-vs, 2020-05-12:
oh got it, debian is awfully behind as usual. thanks!

electrified, 2020-05-13:
It looks like Devuan is based on Debian 9, hence the 4.9LTS kernel. Debian stable (10) has kernel 4.19 so would be OK.

All I can recommend is upgrading your kernel.

closed 2020-05-13 by electrified