electrified/asus-wmi-sensors #66: Weird values (since the beginning)

thagabe, 2020-05-11:
Hello again,

Since getting support for the zenith alpha merged back a year ago, the voltages seem completely weird! Here is an example:
```
asuswmisensors-isa-0000
Adapter: ISA adapter
CPU Core Voltage:               2.40 V  
CPU SOC Voltage:                1.98 V  
DRAM AB Voltage:                2.83 V  
DRAM CD Voltage:                2.86 V  
1.8V PLL Voltage:               3.53 V  
+12V Voltage:                  14.45 V  
+5V Voltage:                    4.99 V  
3VSB Voltage:                   3.33 V  
VBAT Voltage:                   3.05 V  
AVCC3 Voltage:                  3.33 V  
SB 1.05V Voltage:               1.07 V  
CPU Core Voltage:               1.21 V  
CPU SOC Voltage:                1.00 V  
DRAM AB Voltage:                1.45 V  
DRAM CD Voltage:                1.45 V  
CPU Fan:                      1147 RPM
Chassis Fan 1:                   0 RPM
Chassis Fan 2:                   0 RPM
HAMP Fan:                        0 RPM
Water Pump 1:                 2860 RPM
CPU OPT:                      1162 RPM
Water Flow:                      0 RPM
Waterblock Flow:                 0 RPM
EXT Fan 1:                     706 RPM
EXT Fan 2:                     724 RPM
EXT Fan 3:                     728 RPM
EXT Fan 4:                     738 RPM
EXT Fan 5:                    1148 RPM
EXT Fan 6:                       0 RPM
HS Fan:                          0 RPM
CPU Temperature:               +37.0°C  
CPU Socket Temperature:        +38.0°C  
Motherboard Temperature:       +33.0°C  
Chipset Temperature:           +54.0°C  
Tsensor 1 Temperature:        +216.0°C  
CPU VRM Temperature:           +42.0°C  
Water In:                     +216.0°C  
Water Out:                    +216.0°C  
Waterblock In:                +216.0°C  
Waterblock Out:               +216.0°C  
EXT Tsensor 1:                +216.0°C  
EXT Tsensor 2:                +216.0°C  
EXT Tsensor 3:                +216.0°C  
Tsensor 2 Temperature:        +216.0°C  
DIMM.2 Tsensor 1 Temperature: +216.0°C  
DIMM.2 Tsensor 2 Temperature: +216.0°C  
CPU VRM Output Current:        24.00 A
```

as you can see the fan and temps look good. But Voltage and Amps look way off. 

PS: Any chance this driver can get merged into the mainline kernel?

electrified, 2020-05-11:
Hello,

Regarding the incorrect values, it sounds like the WMI is reporting the wrong values (in your first issue you said the values are off in HWiNFO too). This is the case for some other boards too such as the X470 Prime.

I don't want to add board specific hacks to the driver, so I would recommend tweaking the values with compute statements in the sensors config file, there are some examples: https://github.com/lm-sensors/lm-sensors/blob/master/etc/sensors.conf.eg

You can add offsets, scale the values etc.

A more long term approach would be contacting ASUS support and getting them to fix their BIOS, could be worth going via HWiNFO/AIDA64 as they have contacts at ASUS.

Re the mainlining, I'm unsure about it. There are at a minimum a few changes that I know would be needed (at the moment it pretends to be a platform device, so no changes were needed to lm_sensors. That's why it shows up as an ISA device. It should really be a wmi_device, but lm_sensors changes would be needed). There are also issues with WMI access causing fans to stop on some BIOS revisions (fairly catastrophic), so a more thorough whitelist/blacklist based on BIOS versions would be needed.

thagabe, 2020-05-11:
I'll try to get in touch with someone from HWiNFO or AIDA64, sadly the platform is pretty much dead so I'm doubtful they'll even bother. 

Mainlining is always difficult but I'd be a great tool to be baked into every kernel!

closed 2020-05-11 by thagabe

thagabe, 2020-05-25:
Can this be a replacement?

https://phoronix.com/scan.php?page=news_item&px=AMD-Energy-Driver-Linux

KeithMyers, 2020-05-25:
That is just for power readings on the cores and CCX.  Doesn't handle anything from the SIO sensors.