electrified/asus-wmi-sensors #68: Support Gigabyte Aorus Xtreme TRX40?

bertelschmitt, 2020-05-12:
Thank you for this invaluable service. Hopefully, this message can entice you to extend this project to the Gigabyte Aorus Xtreme TRX40. On this top-of the line Threadripper 3 board, LM-sensors finds only the PCI adapter, and it gives only two highly suspicious readings, namely Tdie and Tctl, which are identical (!!!!) I asked Gigabyte repeatedly for technical data on their sensors, but Gigabyte refuses to disclose them, saying “we can’t teach you because specifications have not been published.” 

The board responds splendidly to both HWiNFO and SIV, both deliver highly granular data under Windows, so I am hoping the WMI will be accessible to Linux.

Dmidecode responds as follows:

```
# dmidecode 3.1
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.
# SMBIOS implementations newer than version 3.1.1 are not
# fully supported by this version of dmidecode.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: Gigabyte Technology Co., Ltd.
	Product Name: TRX40 AORUS XTREME
	Version: x.x
	Serial Number: Default string
	Asset Tag: Default string
	Features:
		Board is a hosting board
		Board is removable
		Board is replaceable
	Location In Chassis: Default string
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x0004, DMI type 10, 6 bytes
On Board Device Information
	Type: Video
	Status: Enabled
	Description:    To Be Filled By O.E.M.

Handle 0x003F, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: Onboard LAN RTL8111
	Type: Ethernet
	Status: Enabled
	Type Instance: 1
	Bus Address: 0000:47:00.0

Handle 0x0040, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: X570/590 SATA0
	Type: SATA Controller
	Status: Enabled
	Type Instance: 5
	Bus Address: 0000:4a:00.0

Handle 0x0041, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: X570/590 SATA1
	Type: SATA Controller
	Status: Enabled
	Type Instance: 6
	Bus Address: 0000:4b:00.0 
```

Thank you!

electrified, 2020-05-12:
Hi,

The WMI sensors implementation is an Asus custom API that they've implemented in their firmware - it's not on other manufacturer's boards, so adding support would not be possible.

In terms of what sensors IC you have, it's the ITE IT8688E: https://www.ixbt.com/platform/gigabyte-trx40-aorus-xtreme-review.html

This is the same as used on the Asus boards, and it's troublesome to use directly under linux. (No datasheets mean it's difficult to write a 100% working driver). The unmaintained out of tree it87 driver is your best bet: https://github.com/a1wong/it87

The Tdie/Tctl readings you are seeing are from the k10temp driver I'm guessing? These are read directly from the CPU and would be the same regardless of motherboard. They are the same on some CPUs: https://forums.gentoo.org/viewtopic-p-8347924.html#8347924

closed 2020-05-12 by electrified

bertelschmitt, 2020-05-13:
Thank you for your help. As for Tdie/Tctl, these readings are from a Threadripper 3970X, and Tctl should differ from Tdie by an offset, I believe it is 27 degree centigrade

electrified, 2020-05-13:
Pass - not sure about offsets!

There is some output from a 3970X here and that is showing the same value for both: https://patchwork.kernel.org/patch/11349263/

Also there is no offset defined in the driver for 3rd gen Threadrippers: https://github.com/torvalds/linux/blob/master/drivers/hwmon/k10temp.c#L119

bertelschmitt, 2020-05-13:
As for the offsets, that issue was a source of constant confusion w/ 1st and 2nd gen Threadrippers, and I hope your are right , and the issue is moot.  Anyway, with your help I compiled the  it87  at [https://github.com/a1wong/it87](https://github.com/a1wong/it87), forced the  driver with modprobe to use  ID 0x8733, where lm-sensors found a chip id didn't recognize, and now, at least I get something:
```
k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +44.1°C  (high = +70.0°C)
Tctl:         +44.1°C

it8792-isa-0a60
Adapter: ISA adapter
in0:          +0.95 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +1.49 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +0.98 V  (min =  +0.00 V, max =  +2.78 V)
in3:          +0.40 V  (min =  +0.00 V, max =  +2.78 V)
in4:          +1.78 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.50 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +3.33 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:         +3.25 V
fan1:         790 RPM  (min =    0 RPM)
fan2:        1104 RPM  (min =    0 RPM)
fan3:         715 RPM  (min =    0 RPM)
temp1:        +50.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +40.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +46.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
```
Don't know how reliable the info is, but it's a start. The temp1 - temp3 could be the wired probes that came with the mobo, I stuck one one the cold-plate, and the other on the VRM. The third could be the noise sensor they supplied.  Again, thank your for showing me the way.