electrified/asus-wmi-sensors #69: Waterflow for Asus Crosshair VII shows flow rate in rpm not L/M as in the bios.

Ariesdragon, 2020-05-26:
I install a Koolance INS-FM 19 Flow Meter with a Koolance ADT-FM03 Flow meter frequency adapter.  
With an AMD Ryzen R7 3800x
My motherboard bios version is 3004.
lm-sensors show Water Flow in RPM not L/M as in the BIOS. 
Not sure if its this driver (or even part of the library) or the BIOS.
I check your code to the best of my ability (my C/C++ is rusty).
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

asuswmisensors-isa-0000
Adapter: ISA adapter
CPU Core Voltage:         +1.18 V  
CPU SOC Voltage:          +1.07 V  
DRAM Voltage:             +1.34 V  
VDDP Voltage:             +0.18 V  
1.8V PLL Voltage:         +2.09 V  
+12V Voltage:            +11.88 V  
+5V Voltage:              +4.93 V  
3VSB Voltage:             +3.31 V  
VBAT Voltage:             +3.21 V  
AVCC3 Voltage:            +3.33 V  
SB 1.05V Voltage:         +1.07 V  
CPU Core Voltage:         +1.20 V  
CPU SOC Voltage:          +1.08 V  
DRAM Voltage:             +1.36 V  
CPU Fan:                  737 RPM
Chassis Fan 1:              0 RPM
Chassis Fan 2:            532 RPM
Chassis Fan 3:            553 RPM
HAMP Fan:                2351 RPM
Water Pump:                 0 RPM
CPU OPT:                    0 RPM
Water Flow:              2530 RPM <-Should be in L/M
AIO Pump:                   0 RPM
CPU Temperature:          +43.0°C  
CPU Socket Temperature:   +35.0°C  
Motherboard Temperature:  +33.0°C  
Chipset Temperature:      +59.0°C  
Tsensor 1 Temperature:    +29.0°C  
CPU VRM Temperature:      +36.0°C  
Water In:                 +30.0°C  
Water Out:                +30.0°C  
CPU VRM Output Current:   +3.00 A  

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +43.0°C  (high = +70.0°C)
Tctl:         +43.0°C  

Not sure if it can be fixed.

KeithMyers, 2020-05-26:
I don't think it can be fixed. The water flow header is just a simple fan rpm output that gets massaged into the l/h value in the BIOS.  But that massaged displayed value does not get exported as a WMI value as far as I can tell.  It ends up being rpm.

Which isn't correct anyway based on my external measurement of actual flow rate in l/h.

electrified, 2020-05-27:
Hi there.

The linux kernel HWMON subsystem doesn't have flow rate as a sensor type.

This is all it has: https://github.com/torvalds/linux/blob/master/include/linux/hwmon.h#L20

So showing the value as L/m would not be possible without changes to the kernel (and lm_sensors, and other apps that use the sensor values)

To convert the RPM pulses to a flow rate, you can find out how many pulses your sensor produces per litre and do the maths in the sensor config: https://github.com/lm-sensors/lm-sensors/blob/master/etc/sensors.conf.default. The units will still show as RPM due to the above issue though.

electrified, 2020-05-27:
I've just read the spec of the frequency adapter - https://koolance.com/adt-fm03-flow-meter-frequency-adapter, it says it is already doing the conversion from RPM to L/m.

> The flow meter RPM reading is intended to allow fan detection software to enable the appropriate alarm or shutdown procedure if the flow meter impeller stops rotating. The "RPMs" represent approximate coolant flow rate in mL/min and not the flow meter's revolutions.

The value you are getting is clearly too high for litres per minute. What value is shown in the BIOS?

Ariesdragon, 2020-05-27:
> I've just read the spec of the frequency adapter - https://koolance.com/adt-fm03-flow-meter-frequency-adapter, it says it is already doing the conversion from RPM to L/m.
>
>
> The value you are getting [...]

The BOIS show me a value in L/M not RPM.  So the BIOS must be doing a conversion or it reading the physical Water_Flow header on the motherboard correctly.  I am not sure what is going on with lm_sensors reading it as an RPM and not as L/M.

electrified, 2020-05-27:
My question was what is the actual value?

electrified, 2020-05-27:
i.e. 7L/m or something. FWIW I have a watercooling loop and am interested in getting this working.

We won't be able to fix the units being wrong, but can maybe get the number correct.

Ariesdragon, 2020-05-27:
> My question was what is the actual value?

BIOS reports 81L/M

electrified, 2020-05-27:
OK. 81 L/M is far too high - not a sensible value.

For example a D5 does 1500L per HOUR (https://www.ekwb.com/shop/ek-d5-pwm-g2-motor-12v-dc-pwm-mo-sleeved) , which equates to 25 L/m and that is a best case scenario with a loop with zero restrictions. It will be far lower with tubing and blocks etc.

You can see the expected range of real values as the meter is only rated to 15L/m (https://koolance.com/ins-fm19-coolant-flow-meter)

From what I've read the Asus W_FLOW header is designed to work with the pulse rate output directly from a meter.

What you are seeing in the BIOS with the 81L/m is the RPM to L/m conversion being done twice, once in the Koolance adapter and once in the BIOS

My advice would be to connect it to a normal fan header not the W_FLOW one if you want to use the ADT-FM03 box

Ariesdragon, 2020-05-27:
> OK. 81 L/M is far too high - not a sensible value.
>
> For example a D5 does 1500L per HOUR (https://www.ekwb.com/shop/ek-d5-pwm-g2-motor-12v-dc-pwm-mo-sleeved) , which equates to 25 L/m and that is a [...]

Figure it out.  Both hwmon reads the fan and water flow headers same value regardless of the BIOS. Since I am running a D5 pump (EKWB custom loop) and the conversion is done automatically by the ADT-FM03 in mL/min not L/min all I had to do is divide by 1000.
What I have setup is Conky with a Lua script to calling sensors and displaying that output from an array.  I have setup to only display specific sensor values via a second array. A form of indirect indexing.

electrified, 2020-05-27:
You can ignore sensor values in the config file then they won't be returned
e.g `ignore fan7`

and you can do the /1000  with a compute statement, e.g. `compute temp2 (@*15/43)`

See https://github.com/lm-sensors/lm-sensors/blob/master/etc/sensors.conf.eg

closed 2020-05-27 by electrified