electrified/asus-wmi-sensors #71: Unsupported board: B550M-PLUS

jfdsmit, 2020-06-29:
Output from dmidecode:

```# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: TUF GAMING B550M-PLUS
	Version: Rev X.0x
	Serial Number: <beep>
	Asset Tag: Default string
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Default string
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x0022, DMI type 10, 6 bytes
On Board Device Information
	Type: Video
	Status: Enabled
	Description:    To Be Filled By O.E.M.

Handle 0x0028, DMI type 41, 11 bytes
Onboard Device
	Reference Designation:  Onboard IGD
	Type: Video
	Status: Enabled
	Type Instance: 1
	Bus Address: 0000:00:02.0
```

I see you keep an explicit list of supported boards - how safe/unsafe is it to add my board to the list and rebuild the driver?

electrified, 2020-06-29:
Hi there,

There is no danger adding it, but I don't think it will work. Looking at a photo, your board has the nuvoton NCT6798D, and on the X570/TRX40 baords with the same chipset, Asus have dropped the WMI interface. (Background: The WMI interface was a workaround for the ITE sensors IC causing lockups)

It's most likely that your board also doesn't have the WMI interface. 

Have you tried `sudo sensors-detect --auto` to see if that finds the nuvoton NCT6798D? On the X570 boards it's working/supported by the mainline nct6775 driver.

jfdsmit, 2020-06-29:
You're probably right, sorry for the noise. I did run `sensors-detect` and after it had no noticeable effect, I ended up finding this project. However, on closer inspection, there is indeed an nct6775 compatible chip detected, and the module is loaded. However, nothing shows up when I do `sensors`. I can find this in journalctl:
```
kernel: nct6775: Found NCT6798D or compatible chip at 0x2e:0x290
kernel: ACPI Warning: SystemIO range 0x0000000000000295-0x0000000000000296 conflicts with OpRegion 0x0000000000000290-0x0000000000000299 (\AMW0.SHWM) (20190816/utaddress-204)
kernel: ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver```

Some further research suggests using `acpi_enforce_resources=lax` in GRUB (potentially dangerous and not recommended) or trying the asus_atk0110 driver to go through ACPI, but that has no effect.

electrified, 2020-06-29:
It looks like `acpi_enforce_resources=lax` is needed on your board for the nct driver. As you say, it may cause crashes but many people report no issues.

If you want to conclusively prove if your board has the WMI interface and if it could be supported by this driver, you can try querying it under Windows using Powershell: https://maidavale.org/blog/investigating-asus-wmi-sensors-from-powershell/ (see if the ASUSHW class exists, and if it has the sensors methods)

Alternatively you can open the UEFI image in UEFITool and see if the text `SENX` is present
https://github.com/electrified/asus-wmi-sensors/issues/11#issuecomment-515903663

jfdsmit, 2020-06-29:
Confirmed, no SENX in the BIOS image. I don't have a Windows install on this machine, and I don't plan to have one.

Thanks for the help!

closed 2020-06-29 by jfdsmit