electrified/asus-wmi-sensors #72: ASUS ROG STRIX Z490-E GAMING - Not supported?

mrronchi, 2020-08-12:
I'm trying to get sensor readouts in my archlinux installation but can't seem to get anyhting to work.

Here is my setup:
- `Linux archlinux 5.4.55-1-lts #1 SMP Fri, 31 Jul 2020 17:38:35 +0000 x86_64 GNU/Linux`
- `KDE plasma 5.19.4`
- `Motherboard: ASUS ROG STRIX Z490-E GAMING`

Here is the output of the command `sensors`:
```
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +38.0°C  

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  (crit = +119.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +33.0°C  (high = +82.0°C, crit = +100.0°C)
Core 0:        +32.0°C  (high = +82.0°C, crit = +100.0°C)
Core 1:        +31.0°C  (high = +82.0°C, crit = +100.0°C)
Core 2:        +31.0°C  (high = +82.0°C, crit = +100.0°C)
Core 3:        +32.0°C  (high = +82.0°C, crit = +100.0°C)
Core 4:        +32.0°C  (high = +82.0°C, crit = +100.0°C)
Core 5:        +31.0°C  (high = +82.0°C, crit = +100.0°C)
Core 6:        +31.0°C  (high = +82.0°C, crit = +100.0°C)
Core 7:        +34.0°C  (high = +82.0°C, crit = +100.0°C)

```

I downloaded the aur package from https://aur.archlinux.org/packages/asus-wmi-sensors-dkms-git/ and installed it regularly.
The output of `dkms status` is:
```
asus-wmi-sensors, 57.d3b68db, 5.4.55-1-lts, x86_64: installed
```
However when I reboot and try looking for sensors again I still can't find them.

I tried `sudo modprobe asus_wmi_sensors` but get the following error:
```
modprobe: ERROR: could not insert 'asus_wmi_sensors': No such device
```

and by reading https://github.com/electrified/asus-wmi-sensors/issues/21 I looked at the output of `dmesg`:
```
[  143.041388] asuswmisensors: Vendor: ASUSTeK COMPUTER INC. Board: ROG STRIX Z490-E GAMING BIOS version: 0403 WMI version: 0
[  143.041389] asuswmisensors: Unsupported board
```

and it seems that my board might not be supported, is this correct? What should I do? 

Thanks a lot

KeithMyers, 2020-08-12:
Look at the main page of this repo.  It lists the supported boards.  If you don't have a WMI BIOS it is not supported.  AFAIK, no Intel boards have ever had a WMI BIOS.
[https://github.com/electrified/asus-wmi-sensors](url)

mrronchi, 2020-08-13:
Thanks for your prompt reply and pointer, I didn't realize that. Could you provide some more information on how to do that? Or maybe do you have any other suggestions on how to get sensor readings for my motherboard? Thanks a lot again.

KeithMyers, 2020-08-13:
MSRtools is standard package in Linux distros to unlock the Machine Specific Registers.  Those allow low level access to core clocks, speeds, power usage and core voltages.  i7Z is just a simple command line utility to view those values once they have access to the msr registers.
`sudo apt install msr-tools i7z`
`sudo modprobe msr`

i7z in a command line utility that will show a lot more information than just the standard coretemp driver.
To get better sensor readings you need to install the sensor driver for your SIO chip used on your motherboard.  I don't know specifically which SIO chip your board uses.  But I would hazard a guess that the standard nct6775 driver will probably be it and will work.  You can check for sure by installing the lm-sensors package.
`sudo apt install lm-sensors`

Then run the `sensors-detect` command in the terminal and follow the prompts.  You are looking for what the sensors-detect package finds for your SIO chip on your motherboard.  If it finds it, it will automatically offer to install it for you. But often the sensors-detect does NOT automatically install the driver, but does offer hints as to what SIO chip it found at a ISA address.  If it says if found an nct chip interface try and install the standard distro provided nct6775 driver which covers a host of actual nct hardware.
`sudo modprobe nct6775`

But there is an issue usually with the nct6775 driver because dmesg complains that the acpi address is already occupied.  You can get around that by putting `acpi_enforce_resources=lax` into the kernel command line in grub
`cd /etc/default`
`sudo nano grub`
and editing the GRUB_CMDLINE_LINUX_DEFAULT="quiet splash" and adding this: acpi_enforce_resources=lax so that it reads like this:
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash acpi_enforce_resources=lax"

If you are successful in loading the msr driver and the SIO driver for your motherboard monitoring chip you should load the drivers permanently so they load every time you boot.
`cd /etc/modules-load.d`
`sudo nano modules.conf`
and add msr and the chip driver to the file and save the file. For example:
msr
nct6775
You can prove the modules are loaded by checking with `lsmod` in the terminal which lists all the modules loaded by the system every time you boot.

Some other sensor desktop monitoring utilities that are useful is GKrellm and Conky
`sudo apt install gkrellm` or `sudo apt install conky`
I like the GKrellm interface on the desktop as it is very compact.

electrified, 2020-08-13:
As per Keith's comments, Asus WMI sensors only exists on previous gen Ryzen boards, which had a flaky ITE sensors chip.

Your board has a Nuvoton sensors IC (see https://www.techpowerup.com/review/asus-rog-strix-z490-e-gaming/7.html) so should be supported by the in tree `nct6775` driver.

Closing as it won't be possible to support your board.

closed 2020-08-13 by electrified