electrified/asus-wmi-sensors #74: Strix B450-F Gaming

Swallowtail23, 2020-08-31:
Hi folks, thank you for this -very grateful and it works well! B450F-Gaming with Ryzen5 3600 under CentOS 8 (headless server).

Couple of minor issues to call out:

Output is as follows-
asuswmisensors-isa-0000
Adapter: ISA adapter
CPU Core Voltage:         +1.04 V
VPP MEM Voltage:          +2.51 V
+12V Voltage:             +9.97 V
+5V Voltage:              +4.99 V
3VSB Voltage:             +3.36 V
VBAT Voltage:             +3.25 V
AVCC3 Voltage:            +3.36 V
SB 1.05V Voltage:         +1.05 V
CPU Core Voltage:         +0.00 V
CPU SOC Voltage:          +0.00 V
CPU Fan:                 1326 RPM
Chassis Fan 1:            639 RPM
Chassis Fan 2:            391 RPM
Chassis Fan 3:            675 RPM
AIO Pump:                   0 RPM
Water Pump:                 0 RPM
CPU OPT:                    0 RPM
CPU Temperature:          +39.0°C
CPU Socket Temperature:   +37.0°C
Motherboard Temperature:  +33.0°C
Chipset Temperature:      +46.0°C
Tsensor 1 Temperature:   +216.0°C
CPU VRM Temperature:       +0.0°C
CPU VRM Output Current:   +0.00 A

1. +12V reads at <10V, which is (I hope!) incorrect
2. First CPU Core Voltage reads correctly, but there is a second one (in8) which always reads zero
3. CPU SOC Voltage (in9) always reads zero
4. Tsensor temp (temp5) always reads 216° - it's not got a sensor connected
5. CPU VRM temp (temp6) always zero
6. CPU VRM Output Current always zero, and I cannot seem to "ignore" it in a config file, I can't find its identifier (e.g. "ignore in1" etc.)

I can't run up Windows on this to use hwinfo unfortunately, not without taking out the services the server provides :(

KeithMyers, 2020-08-31:
Item #4 That is the correct value for an unterminated input on the temp sensor headers on a motherboard. To get a valid reading plug in a standard 10Kohm thermistor.
Item #6 To get the proper sensor name for a board sensor for configuration files, use the sensors -u command to get the raw values and the raw sensor names.9

Swallowtail23, 2020-08-31:
> Item #4 That is the correct value for an unterminated input on the temp sensor headers on a motherboard. To get a valid reading plug in a standard 10Kohm thermistor.
> Item #6 To get the proper sensor [...]

For #4 - gotcha, thank you.
For #6 - perfect, it is at curr1 - now ignored as it reads zero all of the time!

Next will be to try and set some ALERT thresholds.

Anyone know why 12V is reading 10V?

KeithMyers, 2020-08-31:
Incorrect formula or incorrectly implemented measuring circuit in the SIO chip.  Measure the actual +12V on the board and then apply a configuration file correction to match the measured voltage.

electrified, 2020-08-31:
As Keith days, it's a bios issue that Asus have not fixed. See
https://www.hwinfo.com/forum/threads/low-readings-on-the-12v-line.6007/

electrified, 2022-02-16:
Closing as not a driver issue