electrified/asus-wmi-sensors #75: ASUS Crosshair VI Hero reboots or locks

gnusuari0, 2020-09-06:
Reading out the sensors constantly with an extension like GNOME Vitals hangs this board. I have it configured to read sensors every 5 seconds. By blacklisting the kernel module asus-wmi-sensors the motherboard stops crashing.
I've seen this issue in other motherboards with WMI interface. I also suffered the issue of fans getting stuck or turned off, which was corrected with a new version of the UEFI. But this crashing has not been solved. Their code base seems to be broken.

I leave this bug report for reference.

PS: It's remarkable that ASUS switched to a different super I/O chip and removed the WMI interface in 500 chipset series.

KeithMyers, 2020-09-06:
It was necessary for ASUS to develop the WMI interface because of their poor decision to use the ITE87 SIO chipset.  Nothing more.  The headaches that hardware caused was avoided in the X570 hardware.

electrified, 2020-09-06:
Hey there, agreed this sounds like a firmware issue.

Are you using a Zen 2 (3xxx) CPU?

gnusuari0, 2020-09-06:
I'm using a Zen CPU (1xxx).

> Hey there, agreed this sounds like a firmware issue.
> 
> Are you using a Zen 2 (3xxx) CPU?

0lvin, 2022-11-13:
@gnusuari0 Do you have access to 'CROSSHAIR VI HERO'? 

Could you please share your dmesg? 
I like to check that ITE87 sensors have conflicts same as NCT6775 sensors in https://bugzilla.kernel.org/show_bug.cgi?id=204807.

Also could you check https://github.com/zeule/asus-ec-sensors, UEFI dump looks as board has methods required for EC.

gnusuari0, 2023-01-07:
Nope, I no longer have it. The issue was related to the Linux bug affecting Zen and Zen+ to a lesser extent: https://bugzilla.kernel.org/show_bug.cgi?id=196683

I ultimately found that enabling Typical Power Idle in the UEFI stopped crashes. But that's a  workaround. The real solution lies in the kernel and it hasn't been commited.

> @gnusuari0 Do you have access to 'CROSSHAIR VI HERO'?
> 
> Could you please share your dmesg? I like to check that ITE87 sensors have conflicts same as NCT6775 sensors in https://bugzilla.kernel.org/show_bug.cgi?id=204807.
> 
> Also could you please check steps described in https://github.com/zeule/asus-ec-sensors, UEFI dump looks as board has methods required for EC.
> 
> Thank you!

closed 2023-01-07 by gnusuari0