electrified/asus-wmi-sensors #78: Patch upstream

0lvin, 2021-10-05:
Could anyone help with patch test before send to linux kernel upstream? Code is based on current repository.  Thanks @electrified for your code. 

https://bugzilla.kernel.org/show_bug.cgi?id=204807#c128   

We need help with check:
+	"ROG CROSSHAIR VII HERO (WI-FI)",
+	"ROG CROSSHAIR VII HERO",
+	"ROG CROSSHAIR VI HERO (WI-FI AC)",
+	"CROSSHAIR VI HERO",
+	"ROG CROSSHAIR VI EXTREME",
+	"ROG ZENITH EXTREME",
+	"ROG ZENITH EXTREME ALPHA",
+	"PRIME X399-A",
+	"PRIME X470-PRO",
+	"ROG STRIX X399-E GAMING",
+	"ROG STRIX B450-E GAMING",
+	"ROG STRIX B450-F GAMING",
+	"ROG STRIX B450-I GAMING",
+	"ROG STRIX X470-I GAMING",
+	"ROG STRIX X470-F GAMING",

Thank you!

electrified, 2021-10-05:
Hey, many thanks for attempting to get my module upstream @0lvin!

I have a `ROG CROSSHAIR VII HERO (WI-FI)`, so can have a go at testing it.

Do you have some simple instructions on which patches to apply etc, and what testing would you like me to do (is `sensors` output working adequate?) 

Thanks

electrified, 2021-10-05:
Hi, I've given it a go and it's working for me :)

0lvin, 2021-10-05:
Thank you :-) 

I have created updated patch: https://bugzilla.kernel.org/attachment.cgi?id=299111

Next review round will be with your patch.

KeithMyers, 2021-10-05:
So do you need someone to test the C7H board also?  How to apply the patch? Already using the asus_wmi_sensors out of tree module.

0lvin, 2021-10-31:
@KeithMyers @electrified Could you check patch from https://lkml.org/lkml/2021/10/31/210 ?

KeithMyers, 2021-10-31:
I'll try and get to this today or tomorrow.

electrified, 2021-11-01:
Hi @0lvin, I will also try and give this a go today

electrified, 2021-11-01:
To get the patch to apply cleanly I needed to apply both of them.

@KeithMyers These were the steps I took on Arch

```
git clone git://git.kernel.org/pub/scm/linux/kernel/git/groeck/linux-staging.git --branch hwmon-next --single-branch --depth 1
cd linux-staging
patch -i ~/Downloads/1.patch -p1
patch -i ~/Downloads/2.patch -p1
zcat /proc/config.gz > .config
make nconfig # enable "device drivers -> hwmon -> Asus X370/470" as a module
make -j16
make modules
sudo make modules_install
sudo cp -v arch/x86_64/boot/bzImage /boot/vmlinuz-linux-sensortest
sudo mkinitcpio -k 5.15.0-rc4-dirty -g /boot/initramfs-sensortest.img
```

electrified, 2021-11-01:
@0lvin 

Nice work, it's all still working fine for me, although haven't had it running for long:

```
ed@zoomer ~/dev/3rdparty/linux-staging (git)-[hwmon-next] % sensors

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +57.4°C  
Tdie:         +47.4°C  

asus_wmi_sensors-virtual-0
Adapter: Virtual device
CPU Core Voltage:          1.42 V  
CPU SOC Voltage:         807.00 mV 
DRAM Voltage:              1.19 V  
VDDP Voltage:            316.00 mV 
1.8V PLL Voltage:          1.85 V  
+12V Voltage:             11.72 V  
+5V Voltage:               4.99 V  
3VSB Voltage:              3.36 V  
VBAT Voltage:              3.23 V  
AVCC3 Voltage:             3.36 V  
SB 1.05V Voltage:          1.07 V  
CPU Core Voltage:          1.37 V  
CPU SOC Voltage:         816.00 mV 
DRAM Voltage:              1.21 V  
CPU Fan:                  816 RPM
Chassis Fan 1:              0 RPM
Chassis Fan 2:            967 RPM
Chassis Fan 3:            948 RPM
HAMP Fan:                 909 RPM
Water Pump:                 0 RPM
CPU OPT:                  789 RPM
Water Flow:               486 RPM
AIO Pump:                1370 RPM
CPU Temperature:          +47.0°C  
CPU Socket Temperature:   +34.0°C  
Motherboard Temperature:  +33.0°C  
Chipset Temperature:      +47.0°C  
Tsensor 1 Temperature:   +216.0°C  
CPU VRM Temperature:      +40.0°C  
Water In:                +216.0°C  
Water Out:                +27.0°C  
CPU VRM Output Current:   41.00 A  

ed@zoomer ~/dev/3rdparty/linux-staging (git)-[hwmon-next] %  uname -a
Linux zoomer 5.15.0-rc4-dirty #2 SMP PREEMPT Mon Nov 1 18:43:33 GMT 2021 x86_64 GNU/Linux
```

KeithMyers, 2021-11-01:
> To get the patch to apply cleanly I needed to apply both of them.

@electrified Could you explain where the "other patch" is located?
The 1.patch just errored out.  Assume the "other patch" is needed as a precursor.
Went back up the commit thread and couldn't find any reference to it.

electrified, 2021-11-02:
> > To get the patch to apply cleanly I needed to apply both of them.
> 
> @electrified Could you explain where the "other patch" is located? The 1.patch just errored out. Assume the "other patch" is needed as a precursor. Went back up the commit thread and couldn't find any reference to it.

Apply the patch 1/2 from https://lkml.org/lkml/2021/10/31/209
and then the 2/2 from https://lkml.org/lkml/2021/10/31/210

deri971, 2021-11-07:
please add ROG STRIX Z390-F GAMING

electrified, 2021-11-16:
> please add ROG STRIX Z390-F GAMING

This board does not have the required WMI interface (checked as per https://github.com/electrified/asus-wmi-sensors/issues/11#issuecomment-515903663)

Generally only 1st and 2nd gen Ryzen boards have it.

0lvin, 2021-11-16:
@electrified I sent patch without Z390, and added you as one of maintainers if you do not mind.  

There is exists possibility of Intel support, look to https://github.com/NadirZenith/asus-wmi-sensors/commit/61e50640aebbf306b3f547141bb64a204cb71ed5. I am not sure what does it mean, maybe it was just test code or some bios versions has some limited support.

electrified, 2021-11-16:
@0lvin Hi Denis, adding me as a maintainer is fine :)

Regarding the Z390 support, yes it's plausible an older BIOS version had it, it's not present in the latest 1903 version though.

Cheers,
Ed

electrified, 2022-03-23:
Hi @0lvin, Many thanks for your work upstreaming this driver!

I'm going to close this issue now.

Thanks,
Ed

closed 2022-03-23 by electrified

electrified, 2022-04-03:
Hi @0lvin,

Have received this bug report about the upstreamed driver - it looks like one of the motherboard names got modified:

https://github.com/electrified/asus-wmi-sensors/pull/79#issuecomment-1086738317