electrified/asus-wmi-sensors #80: Kernel Release 5.17

xuedi, 2022-03-06:
Hello, i did read that there have been lots of improvements for these asus boards in upstream kernel 5.17. (https://www.phoronix.com/scan.php?page=news_item&px=ASUS-EC-Sensor-Driver)

Are this your patches, will they conflict with your DKMS?

Thanks a lot for your amazing work and the Arch package :-)

Cheers
xuedi

electrified, 2022-03-23:
Hi there,

Thanks!

The driver contained in this repository is now part of the main kernel tree as of 5.17, it was upstreamed by Denis Pauk. It has the same name `asus_wmi_sensors`. See https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/hwmon/asus_wmi_sensors.c?h=v5.17

Checking the Arch kernel config, this driver is being built as a module: https://github.com/archlinux/svntogit-packages/blob/packages/linux/trunk/config#L5156 so the AUR package will be no longer be necessary once kernel 5.17 moves out of testing. I will remove the AUR package when that happens.

Regarding the Phoronix article, https://www.phoronix.com/scan.php?page=news_item&px=ASUS-EC-Sensor-Driver the driver mentioned there that is being replaced (`asus_wmi_ec_sensors`) supports a different set of boards, X570 and B550 than the boards supported by this driver (See https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/hwmon/asus_wmi_ec_sensors.c?h=v5.17#n201)
The new driver replacing that (`asus-ec-sensors`) looks to support the same boards, so I don't think that will be replacing `asus_wmi_sensors`

Cheers,
Ed

xuedi, 2022-03-23:
Thanks a lot for this detailed answer :-)

adatum, 2022-05-29:
@electrified Congrats on having the driver included in the kernel! 

Should the dkms `asus-wmi-sensors` driver now be uninstalled if running kernel 5.17+ ?

electrified, 2022-06-14:
Hey @adatum yes the DKMS driver can be uninstalled.